Azure-Samples/iot-middleware-freertos-samples #45: External dependencies have to be recloned whenever you delete the build folder

ryanwinter, 2021-06-21:
### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [x] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
Follow guide steps here: https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/NXP/mimxrt1060#build-the-image

During the configure step, multiple repositories are cloned into the temporary build directory. There are [two disadvantages here](https://github.com/azure-rtos/getting-started/issues/74):
1. The repositories need to be recloned every time the build directory is deleted.
2. Can confuse debugger if files are temporary.

### Expected/desired behavior
External repositories to not be stored in temporary build directories, and to be easy to update when needed.

ryanwinter, 2021-06-30:
https://github.com/azure-rtos/getting-started/issues/74

hihigupt, 2021-07-22:
Our cmake make sure if you have pre-seeded external dependencies, we will not download it to temporary build folder. To make cmake aware about these dependencies, user should set cmake variables with their path e.g.: FREERTOS_PATH  for freertos.

Check this cmake for the logic.
https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/130bc92100f2fa35f4aedff7819f8fa195e6dea7/cmake/common/utilities.cmake#L28

closed 2021-07-22 by hihigupt