Azure-Samples/iot-middleware-freertos-samples #120: Cannot launch VS Code debugger with STM32

ademyankov, 2021-11-12:
Well, basically nothing works out of the box or by just simply following the [instructions](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/projects/ST/b-l475e-iot01a/VSCodeDebug.md):

The first problem is that the PowerShell script simply doesn't work:

[image]

However, I was able to execute the script manually from the CMD and it successfully fetched the RTOS.

If I skip it I hit the problem with openocd:

[image]

I tried to add -s command to openocd:
[image]

I can launch it manually from the power shell:

[image]

OpenOCD folder looks like this:
[image]

PATH variable points to C:\ST\OpenOCD-20210729-0.11.0

I have noticed a weird thing...

[image]

why does it add '-s' and the root folder of the IoT project for the openocd cfg search path? 

I couldn't find any JSON settings explicitly doing this.

After I copied folders from the openocd into the root of the project at least the debugger starts.

**Since you guys went through this hassle can you see or possibly identify where the problem might be with all those things?**

danewalton, 2021-11-12:
Hey @ademyankov thanks for moving this over.

At least for the freertos download, [we have a PR here which should fix that](https://github.com/Azure-Samples/iot-middleware-freertos-samples/pull/118). Looks like your default integrated terminal is `cmd` so it will need the explicit powershell invocation to run it properly.

I have a feeling the `cmd` vs `powershell` idea is also what is causing the other parts to fail which is why the terminal with the blue background you posted (powershell) is working, while running it from VSCode does not. I can look into making sure it works if `cmd` is the default terminal.

ademyankov, 2021-11-12:
@danewalton thanks, man, good to know. Actually, I don’t give a d* about cmd :) if I can make powershell as a default terminal and everything will work I would easily go with that.

Can you share the trick to make it happen?

danewalton, 2021-11-12:
Oh wonderful that helps then :)

If you go to this setting in VSCode you should be able to change it to run Powershell.

[image]

Actually change this one too since this says specifically for "tasks"

[image]

Should look like this

[image]

ademyankov, 2021-11-13:
Ok, I got a bit further. Now, it uses PowerShell, it fetches FreeRTOS, builds the code but still has issues with openocd.

[image]

ademyankov, 2021-11-13:
Just wanted to let you know if I copy all the stuff from openocd: board, interface, and target into the root of the project then the debugger starts and all works as expected.

[image]

danewalton, 2021-11-16:
Okay did a bit of digging. The error message you're getting is

> Error: Can't find board/stm32l4discovery.cfg

[On this page of the docs for OpenOCD](https://elinux.org/OpenOCD_Ubuntu_Package) it says the following

> The package installs config files to /usr/share/openocd/scripts, but OpenOCD looks for config files in /usr/local/share/openocd/scripts. Fix that now by creating a symbolic link in /usr/local/share to /usr/share/openocd. In the terminal window, type:
> ```bash
> cd /usr/local/share
> sudo ln -s /usr/share/openocd .
> ```
> OpenOCD will now be able to find its config files. For more information on how OpenOCD searches for config files, see OpenOCD Config File Paths.

OpenOCD first looks in the "current directory" to find the config file, so when you say you copy the file to the top level, it makes sense it would find it. 

I don't think you're on Linux but it's possible you might be having a similar issue?

ademyankov, 2021-11-16:
Yes, @danewalton, I am not on Linux. I am on Windows 10.

danewalton, 2021-11-16:
The fact that putting the board configuration in the top level allows it to work has me believing some path variable isn't set up correctly. Are you able to see the attached screenshot and the link provided to make sure the OpenOCD environment is set up correctly?

[image]

https://elinux.org/Running_OpenOCD_on_Windows

ademyankov, 2021-11-16:
Yes, I have seen this page... and my windows env var points to openocd folder where all those directories are. But as I pointed it out before, somewhere, some settings changes the search path for openocd tool with '-s' switch.

[image]

As you can see it points to the root of the project.

I tried manually add that switch into the task.json to point it to the right directory:

[image]

But it did not help at all.

I will dig further and if I find a solution I will definetely add a comment here.

ademyankov, 2021-11-16:
Oh, BTW, @danewalton, I have a question.

When the debuger is run it downloads and builds FreeRTOS repo via `.github/scripts/fetch_freertos.ps1` script into **lib/FreeRTOS** folder and then firmware binary is built using this version of FreeRTOS.

The question is what is the reason for doing so? Because, if the project is built without debugging then the FreeRTOS is fetched via cmake script into **b-l475e-iot01a/_deps/freertos-src** folder. So, basically the project already has FreeRTOS fetched and built.