Azure-Samples/iot-middleware-freertos-samples #136: Failed to perform TLS handshake / example running on Linux PC

acvin, 2022-01-19:
### This issue is for a: (mark with an `x`)
```
- [ x] bug report
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression
```

### Minimal steps to reproduce
> By following the guide for a Linux PC, I just run the executable with a  
`sudo ./build_linux/demos/projects/PC/linux/iot-middleware-sample` 

### Any log messages given by the failure
> [INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:162] ---------STARTING DEMO---------
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:171] 
IP Address: 192.168.1.100
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:174] Subnet Mask: 255.255.255.0
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:177] Gateway Address: 192.168.1.1
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:180] DNS Server Address: 1.1.1.1

[INFO] [AzureIoTDemo] [prvConnectToServerWithBackoffRetries:579] Creating a TLS connection to MYHOSTNAME:8883.
[ERROR] [TlsTransport] [tlsHandshake:573] Failed to perform TLS handshake: lMbedtlsError[-29312]= SSL - The connection indicated an EOF : <No-Low-Level-Code>.
[ERROR] [TlsTransport] [TLS_Socket_Connect:727] Failed to do TLS handshake 4.
[WARN] [AzureIoTDemo] [prvConnectToServerWithBackoffRetries:604] Connection to the IoT Hub failed [4]. Retrying connection with backoff and jitter [133]ms.
[INFO] [AzureIoTDemo] [prvConnectToServerWithBackoffRetries:579] Creating a TLS connection to MYHOSTNAME:8883.

> Application continues with multiple retries, with the same error message

### Expected/desired behavior
> Device has been correctly setup with a SAS authentication method and device name, hostname and device primary key have been added in the demos/projects/PC/linux/config/demo_config.h file
> #undef democonfigENABLE_DPS_SAMPLE has been added and democonfigCLIENT_CERTIFICATE_PEM commented out

### OS and Version?
> Running on Linux Debian 10, x86

### Versions
> source code from commit 9e836612f25a06a8ab2da65bac9e9f4d22756a20 (current main branch)

acvin, 2022-01-21:
Hi all, 
as an update, I got the same error even with X509 authentication procedure (in place of the SAS method described above)

danewalton, 2022-01-21:
Hi @acvin 
Thanks for the question. Off the top of my head, I'm hoping to do a few sanity checks. I want to make sure these credentials are formatted correctly.

For SAS key, it should resemble the following:

```c
/**
 * @brief Device symmetric key
 *
 */
#define democonfigDEVICE_SYMMETRIC_KEY      "fsOVd4h4yabcd7CrgnMGP4UGVnPM0NbFRxyzxvbSoC4="
```

For the Cert auth (RSA), it should resemble the following:

```c
/**
 * @brief Client's X509 Certificate.
 *
 */
#define democonfigCLIENT_CERTIFICATE_PEM    \
"-----BEGIN CERTIFICATE-----\n" \
"MIIC/zCCAeegAwIBAgIUUISm0r63y4CCdmo4wjQKFehIaMMwDQYJKoZIhvcNAQEL\n" \
"BQAwDzENMAsGA1UEAwwEdGVtcDAeFw0yMjAxMjExOTE3MzFaFw0yMjAyMjAxOTE3\n" \
"MzFaMA8xDTALBgNVBAMMBHRlbXAwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEK\n" \
"AoIBAQD6cmYAi9EUTQlNTRLXXdx7ialWAEziXyeV1mNnwVBKYD90T1g7pdo+2xRm\n" \
"kaMEonk1PAHSRyXz8XcUdLup2Gzr/C8iOh71pv3mXpLcf0RwQiRDG5DvFU/VMADm\n" \
"ugGCSe2A/PCAqqICQrLWOIToTD3GMwjvvfYvxLbA+RbN3cXfImzDjFQniUduzCgO\n" \
"Z4gxq0TX0FSC3VhJNmo8RTf+SLl/1S5aTQNCzvjcj/juv1NnDfbSd36efNXSLtA2\n" \
"qxGuqEhy/n0+GVBeXUgfJ2Ns7VBng7gBOabh/V23vP8gd8BIIRAEpdVrkWHCVjZo\n" \
"HAqsYrDBLZMPO6Lot8hC7abcdn8lAgMBAAGjUzBRMB0GA1UdDgQWBBRvKB3uNgdB\n" \
"wykTGf3OYzl9n3/nQzAfBgNVHSMEGDAWgBRvKB3uNgdBwykTGf3OYzl9n3/nQzAP\n" \
"BgNVHRMBAf8EBTADAQH/MA0GCSqGSIb3DQEBCwUAA4IBAQCDx/g2xbASB6cwxSAU\n" \
"GQCeDyFR2ZAjJ19zQVn1RPvHi3lhfSZJA++JrpvCKOokbp4p3rRArxsVSELAG9I4\n" \
"DWj0G962ffpJO4FhGRNQAceQVPkSrk/gefhHIE5zBFxkWnA/QoVkmDVslciYdRGx\n" \
"vrit6/p3JUS9HN9nF7pX+Nf4q1cqSZTiQHawVFGogjTOMs8KuaqqObDCz25e7bag\n" \
"SX7Ld+pFa7Za4feyI1UZEkVXsLFwV1hYd24BKwR293iD+gt2EIrUOx3wL0SLIIl5\n" \
"ohd05GMa3KuKCDKGnjEfzJbI87LZusU9p3VuaXyeKhD4IShVLW9xNIU+2CjOkWoN\n" \
"uqcc\n" \
"-----END CERTIFICATE-----"

/**
 * @brief Client's private key.
 * 
 */
#define democonfigCLIENT_PRIVATE_KEY_PEM    \
"-----BEGIN RSA PRIVATE KEY-----\n" \
"MIIEpgIBAAKCAQEA+nJmAIvRFE0JTU0S113ce4mpVgBM4l8nldZjZ8FQSmA/dE9Y\n" \
"O6XaPtsUZpGjBKJ5NTwB0kcl8/F3FHS7qdhs6/wvIjoe9ab95l6S3H9EcEIkQxuQ\n" \
"7xVP1TAA5roBgkntgPzwgKqiAkKy1jiE6Ew9xjMI7732L8S2wPkWzd3F3yJsw4xU\n" \
"J4lHbswoDmeIMatE19BUgt1YSTZqPEU3/ki5f9Uhjk0DQs743I/47r9TZw320nd+\n" \
"nnzV0i7QNqsRrqhIcv59PhlQXl1IHydjbO1QZ4O4ATmm4f1dt7z/IHfASCEQBKXV\n" \
"a5FhwlY2aBwKrGKwwS2TDzui6LfIQuzVAwp/JQIDAQABAoIBAQCryTZQH5ddPm4n\n" \
"+Lr4wOgQUN+3HuOdzrkfXPu4WnARZC2CQT57Q4sM3EcFB9ZywmMzcaKqMqVEy5br\n" \
"kBp9KPelcsh9yLIdu9qiXrKLzjwKaEPpSApFBDBLQe8MU//e94iSlmz9RI110XxL\n" \
"1l7L3CfvQGhb0cgBAgR9Hx5WPU2abcdeLSruO73AO7iPj/ikV+CJr9qxAUU41mES\n" \
"VBUBkjSnkUscVoP1Hb9RNki+oILv/6+kUjn+0O/MBxERH8jR7KECdwYoeaprB0/M\n" \
"BK4e/qdXr8wA+Gyeger5k2/fagPWn6OzOYJPcZ2SAQctXlyvRJxjN4wXG38PQN4c\n" \
"KpXR7GmhAoGBAP06t4No69Vk2Fxpm40x4MdxotUAN6OKvkIDuFlc0TVNM6YNmr0m\n" \
"6McBNUtkAbGz7RNolsj5qx/39kTOmqJJYwcQ/sYA6IGDZZ/lViT8ZBXEqoK3exNc\n" \
"nnO7pCyo91G+rpcYMPp2aij6y0Jm9MhaXy87Cy9rerkwaWlC98J2eSC5AoGBAP0v\n" \
"41HN7LNaZRhuaez0NGimThfajZHs2D2szcWUxCOezgFW+koBn/mWHmAh7nej9DEN\n" \
"HvsUqomsDbl7eo8JqqErX7LeRSaNR60gkcYu/tY+3/Pd+oW+3N6h02TRuzfYgvUC\n" \
"XQIKo+eLduxh7x0Fc1w2XfvjLHKXUP0YELccIiPNAoGBAPSAMxyQu9SI0Fc6KNUF\n" \
"qdEoqV4cHZyso9M9oEu4UM+0QQA99oBYBz3OrtSiNPiNj4TD4YMffdkaJBB/o9K6\n" \
"JQF3qWcpDVWvtvnSB4EiqMhT4Ybg/dbOALgiHvx4EIQ/JngBBZlvqET1EjiFDCg/\n" \
"CdSin8QjdJVGZyEZZfu/RGkRAoGBAJMf98V6lzhlp+d5id4izi2drjWYT7njWokE\n" \
"WJKe7xcP2f/XThwhM5fyTvZn4U3unSC/0ifvC2Zy74rAISjrYhbDwJmNJbvQTz1g\n" \
"drUma/mNkdABRunfTjOk79VfNp6uFvNPC0L7ceFEHrgf9LX1i+KiyRxtF1ADdBGO\n" \
"oW/2TjRlAoGBAJk+pS8PTXPaGmGGP/HS5xfeBr3+iSg2oh+bQ/8SUoY4lBpBUFCl\n" \
"ga+tINVoaaRKs51FYgXfnjEBJug71lfBoeKPyhqvu/wpwHTHcYG7kXjvlc6VdKOj\n" \
"pZUTBPSNyMWddrmykPTEsbnDutgIbt5LMFH2RVc+luX39XOYkJ9i+1R7\n" \
"-----END RSA PRIVATE KEY-----"
```

Do yours look similar?

acvin, 2022-01-24:
Hi @danewalton , 
both look similar. The formatting for SAS and the headers for the X509. I have tried to switch between SAS and X509, same results. It seems related to the Mbedtls issue. 
I will try to fall back to a previous version to see if anything has changed. 
Any hint from your side? Could I activate a more verbose debug?

danewalton, 2022-01-25:
Hmmm I'm running on windows WSL2 with Ubuntu 20.04. I have the samples working on it.

Are you able to get a Wireshark trace of what you are seeing?

acvin, 2022-01-25:
Hi @danewalton, 
I am repeating the operation over WSL2 on Windows 10, using Ubuntu 20.04 LTS. Debug messages are slightly different with the same error message. Please find below my debug console:

`[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:162] ---------STARTING DEMO---------
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:171]
IP Address: 192.168.1.100
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:174] Subnet Mask: 255.255.255.0
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:177] Gateway Address: 192.168.1.1
[INFO] [AzureIoTDemo] [vApplicationIPNetworkEventHook:180] DNS Server Address: 1.1.1.1
[INFO] [AzureIoTDemo] [prvConnectToServerWithBackoffRetries:580] Creating a TLS connection to cuc.azure-devices.net:8883.
[ERROR] [TlsTransport] [tlsHandshake:571] Failed to perform TLS handshake: lMbedtlsError[-128]= <No-High-Level-Code> : <No-Low-Level-Code>.
[ERROR] [TlsTransport] [TLS_Socket_Connect:729] Failed to do TLS handshake 4.
[WARN] [AzureIoTDemo] [prvConnectToServerWithBackoffRetries:603] Connection to the IoT Hub failed [4]. Retrying connection with backoff and jitter [133]ms.`

Please find attached the Wireshark dump. It seems is correctly started, then it gets a RST request from the host. 

[image]
Thank you

acvin, 2022-01-31:
Hi @danewalton 
Finally figured out the issue, that was related to our internal firewall preventing incoming DTLS connections. 
Thank you for your support!

closed 2022-01-31 by acvin

danewalton, 2022-01-31:
Sounds good glad you were able to figure it out!