Azure-Samples/iot-middleware-freertos-samples #137: Cannot build samples

droll80, 2022-01-19:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
Tried to build in PowerSheell and in VS Code terminal
### Any log messages given by the failure
>
```
PS C:\Sources\IoT\iot-middleware-freertos-samples> cmake -G Ninja -DVENDOR=ST -DBOARD=stm32h745i-disco -Bstm32h745i-disco .
-- No STM32_TARGET_TRIPLET specified, using default: arm-none-eabi
-- No FREERTOS_PATH specified: Fetching FreeRTOS to path <>
-- Populating freertos
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Sources/IoT/iot-middleware-freertos-samples/stm32h745i-disco/_deps/freertos-subbuild
[0/7] Performing update step for 'freertos-populate'
[2/7] No patch step for 'freertos-populate'
[3/7] No configure step for 'freertos-populate'
[4/7] No build step for 'freertos-populate'
[5/7] No install step for 'freertos-populate'
[6/7] No test step for 'freertos-populate'
[7/7] Completed 'freertos-populate'
INFOFREERTOS_PATH set to
-- Populating stm32cubeh7
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Sources/IoT/iot-middleware-freertos-samples/stm32h745i-disco/_deps/stm32cubeh7-subbuild
[0/7] Performing update step for 'stm32cubeh7-populate'
[2/7] No patch step for 'stm32cubeh7-populate'
[3/7] No configure step for 'stm32cubeh7-populate'
[4/7] No build step for 'stm32cubeh7-populate'
[5/7] No install step for 'stm32cubeh7-populate'
[6/7] No test step for 'stm32cubeh7-populate'
[7/7] Completed 'stm32cubeh7-populate'
-- Populating lwip
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Sources/IoT/iot-middleware-freertos-samples/stm32h745i-disco/_deps/lwip-subbuild
[0/7] Performing update step for 'lwip-populate'
[2/7] No patch step for 'lwip-populate'
[3/7] No configure step for 'lwip-populate'
[4/7] No build step for 'lwip-populate'
[5/7] No install step for 'lwip-populate'
[6/7] No test step for 'lwip-populate'
[7/7] Completed 'lwip-populate'
INFOLWIP_PATH set to
-- Search for HAL families: STM32H7_M7
-- Search for HAL drivers: adc;can;cec;comp;cortex;crc;dac;dma;exti;flash;gpio;i2c;i2s;irda;iwdg;pcd;pwr;rcc;rtc;smartcard;smbus;spi;tim;tsc;uart;usart;wwdg;eth;hcd;mmc;nand;nor;pccard;sd;sram;cryp;dcmi;hash;rng;hrtim;opamp;sdadc;dfsdm;dma2d;dsi;flash_ramfunc;fmpi2c;lptim;ltdc;qspi;sai;sdram;spdifrx;jpeg;mdios;cordic;fdcan;fmac;dts;gfxmmu;hsem;mdma;ospi;otfdec;pssi;ramecc;swpmi;firewall;lcd;pka;gtzc;icache
-- Search for HAL LL drivers: adc;comp;crc;crs;dac;dma;exti;gpio;i2c;pwr;rcc;rtc;spi;tim;usart;usb;utils;fsmc;sdmmc;rng;fmc;hrtim;opamp;dma2d;lptim;lpuart;ucpd;cordic;fmac;bdma;delayblock;mdma;swpmi;pka

CMake Error at CMakeLists.txt:45 (add_subdirectory):
  The source directory

    C:/Sources/IoT/iot-middleware-freertos-samples/libs/azure-iot-middleware-freertos

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
See also "C:/Sources/IoT/iot-middleware-freertos-samples/stm32h745i-disco/CMakeFiles/CMakeOutput.log".
```
### Expected/desired behavior
>

### OS and Version?
Windows 10
### Versions
```
PS C:\Sources\IoT\iot-middleware-freertos-samples> ninja --version
1.10.2
```
```
PS C:\Sources\IoT\iot-middleware-freertos-samples> cmake --version
cmake version 3.22.1
```

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

danewalton, 2022-01-19:
Ahhh yes this means submodules were not initialized. Use the following from the root directory of the project.

```bash
git submodule update --init --recursive
```

Let me know if that helps!

droll80, 2022-01-20:
This helped! Thanks a lot.

closed 2022-01-20 by droll80