Azure-Samples/iot-middleware-freertos-samples #142: Fail during menuconfig in "Connect an ESPRESSIF ESP32 using Azure IoT Middleware for FreeRTOS" sample.

IPlayZed, 2022-03-01:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)

```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce

> Use a Linux based system, Bash, with ESP-IDF configured and sourced in the current shell, as explained in the official documentation.
> Follow the steps until `idf.py menuconfig` in [Connect an ESPRESSIF ESP32 using Azure IoT Middleware for FreeRTOS READ](https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/aziotkit#connect-an-espressif-esp32-using-azure-iot-middleware-for-freertos)

### Any log messages given by the failure

```
07:59:14 minef@mineflaptoparch aziotkit ±|main|→ idf.py menuconfig
Executing action: menuconfig
Running cmake in directory /home/minef/esp/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/aziotkit/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/minef/esp/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/aziotkit"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.35.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/minef/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/minef/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/minef/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/minef/esp/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/aziotkit/sdkconfig
Loading defaults file /home/minef/esp/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/aziotkit/sdkconfig.defaults...
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /home/minef/.espressif/python_env/idf5.0_py3.10_env/bin/python3.10 (found version "3.10.2") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "azure_iot_freertos_esp32" version: bcfbaab
-- Adding linker script /home/minef/esp/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/aziotkit/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/minef/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/minef/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/minef/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/minef/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/minef/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/minef/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/minef/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/minef/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/minef/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
CMake Error at /home/minef/esp/esp-idf/tools/cmake/component.cmake:304 (message):
  Include directory
  '/home/minef/esp/esp-idf/components/freertos/include/freertos' is not a
  directory.
Call Stack (most recent call first):
  /home/minef/esp/esp-idf/tools/cmake/component.cmake:477 (__component_add_include_dirs)
  components/azure-iot-middleware-freertos/CMakeLists.txt:29 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "/home/minef/esp/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/aziotkit/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

### Expected/desired behavior
>

### OS and Version?
> `Linux mineflaptoparch 5.16.11-arch1-1 #1 SMP PREEMPT Thu, 24 Feb 2022 02:18:20 +0000 x86_64 GNU/Linux`, standard kernel

### Versions
> `ESP-IDF v5.0-dev-1624-gc04803e88b`
> `GNU bash, version 5.1.16(1)-release (x86_64-pc-linux-gnu)`

### Mention any other details that might be useful

The directory `/home/minef/esp/esp-idf/components/freertos/` does exist, but it does not contain an `include` folder.
The EDP-IDF is sourced via the Bash alias `alias get_idf='. $HOME/esp/esp-idf/export.sh'`.

danewalton, 2022-03-01:
Hi @IPlayZed 

We previously had that change here: https://github.com/Azure-Samples/iot-middleware-freertos-samples/pull/119/files

We reverted it [here](https://github.com/Azure-Samples/iot-middleware-freertos-samples/pull/123/files) because we wanted to make sure we were snapped to a particular version of the ESPIDF. We've had crashes on newer versions and support only the versions which are listed in the README to make sure we know scenarios which work.

Going to close this and your PR. Thank you for submitting the change. If you have any other questions please let us know.

closed 2022-03-01 by danewalton

IPlayZed, 2022-03-02:
Hi @danewalton 

I am not sure than what should I do. Maybe I am looking in the wrong place but what ESP-IDF version should I use than for the demo project?
As it stands now it is not possible to config or build it.

danewalton, 2022-03-02:
We only have it confirmed to work with the versions which we list. The stable [v4.4 has the path as we have it](https://github.com/espressif/esp-idf/tree/v4.4/components/freertos/include/freertos). Even the [5.0dev](https://github.com/espressif/esp-idf/tree/v5.0-dev/components/freertos/include) version has it as we do. I think it's only the master which changes it. In the future, if they have a major version change which we can use and confirm it works, then we will update. We unfortunately can't guarantee it will work with any given master version they have.

TLDR: if you want support for the scenario in your project, we can only provide it for versions which we test against.