Azure-Samples/iot-middleware-freertos-samples #144: Azure IoT Hub samples behind firewalls / rtosveth0 and rtosveth1 usage

acvin, 2022-03-04:
```
- [ ] bug report
- [ ] feature request
- [X] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> The setup creates two virtual interfaces, rtosveth0 and rtosveth1, to run simulator on a Linux Debian machine.

### Expected/desired behavior
> Is there a way to run the application to use natively the physical Ethernet interface without passing through the virtual interfaces? The issue I am having is that the DTLS / MQTT packets sent from the machine to the Azure IoT Hub are originated from a 192.168.1.100 on rtosveth1 IP different from the machine IP. The firewall and NAT have troubles to route packets to that location in response. 

### OS and Version?
> Linux Debian 11

danewalton, 2022-03-09:
Hi @acvin 

You might be able to get the index for `eth0` and try that but I haven't personally tried it as I only have WSL on my windows machine. If your box is a native Linux box and has an ethernet port, that might work.

closed 2022-03-17 by danewalton

acvin, 2022-03-31:
Hi @danewalton , it is confirmed working with eth0 without passing through the virtual interfaces