Azure-Samples/iot-middleware-freertos-samples #154: How to integrate the middleware into ESP32 Arduino projects?

IPlayZed, 2022-03-29:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Create an Arduino project and add [ESP32 Arduino](https://github.com/espressif/arduino-esp32) to the Board manager.
> Use the middleware source in the project codebase.

### Expected/desired behavior
> Middleware is available as a custom library.
> Can successfully build and flash the project. 

### OS and Version?
> Linux, preferaebly Arch. (Altough, the problem should be distro agnostic.)

### Versions
> The latest ESP32 Arduino Core.

### Mention any other details that might be useful

I am planning to create an Arduino project in a reproducible and in a "click, build, flash, run" way. Meaning I can provide a repository with and Arduino project and the middleware (preferably as a submodule) with related dependencies. The sample ESP32 project in this repository is a very good example when explicitly using ESP-IDF, but I am a little lost how can I integrate this when using the Arduino IDE and the related Arduino API.

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

wduraes, 2022-03-29:
Hi @IPlayZed the ESP32 is already available to be installed in the Arduino's Board manager as you can see [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html). Our [guide ](https://github.com/Azure/azure-sdk-for-c-arduino/blob/main/examples/Azure_IoT_Central_ESP32/readme.md)to running the ESP32 with Azure also include this step. If you need to see samples using ESP-IDF, then I suggest you check our [middleware for FreeRTOS samples](https://github.com/Azure-Samples/iot-middleware-freertos-samples).

IPlayZed, 2022-03-29:
> Hi @IPlayZed the ESP32 is already available to be installed in the Arduino's Board manager as you can see [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html). Our [...]

Hi, thanks for the answer!

So using the middleware is not recommended when building an Arduino project, if I understand it right, as azure-sdk-for-c-arduino should provide the needed functionality?

wduraes, 2022-03-29:
That's correct: middleware for FreeRTOS samples for the ESP32 were built for use with ESP-IDF. If you're using the Arduino samples for ESP32 they already provide the needed functionality to get your device connected to Azure IoT. You don't need to mix up the two to get a working device.

wduraes, 2022-04-06:
Hi @IPlayZed since this seems to be clarified I'll go ahead and close this issue. Happy to help if you have any other questions in the future!

closed 2022-04-06 by wduraes