Azure-Samples/iot-middleware-freertos-samples #165: ESP32 Azure custom data transmissions

mojy25, 2022-04-20:
### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
Follow ESP32 connection to Azure guidelines with the firmware package (https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/esp32), and a custom ESP32 PCB
Verify connection with Azure
Receiving transmissions from ESP32 board that read:

[image]

### Any log messages given by the failure
>
No failure to report

### Expected/desired behavior
>
I would like to send data collected from the custom ESP32 board (simple voltage readings from an ADC instead of the temperature data. 
I connected the ESP32 to Azure, and the connection is verified with received data, using the no Plug and Play template. The device is currently registered as a thermometer. 
What do I do to send the voltage readings from the ADC to Azure instead of the thermometer data? Is there a seperate device template from the thermometer template that will allow me to send custom data easier? Any sort of guidance in this process is appreciated. 

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Windows 10

### Versions
>
ESP v4.4 updated, April 5th
### Mention any other details that might be useful
The board connects, and I've read the voltage readings with accuracy to what is expected in a custom Python software. The hardware is not an issue here, as we are connected to Azure with no problem. I've transmitted the data over Wifi using sockets from the ESP32 board to a Python software, so the ADC data is accurate over WiFi as well. I'm new to Azure and am looking to transmit the same data over WiFi to Azure instead, and was hoping for some guidance in how to make this happen. 

Thank you for any help

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

mojy25, 2022-04-20:
Current ESP32 log showing connectivity and message publishing to Azure:

[0;32mI (28509) AZ IOT: Successfully acquired IoT Hub name and Device ID[0m
[0;32mI (28529) AZ IOT: Creating a TLS connection to iotc-c6e37316-0ec9-4575-8839-125d7a50f8bc.azure-devices.net:8883.

[0m
[0;32mI (30549) tls_freertos: (Network connection 0x3ffca830) Connection to iotc-c6e37316-0ec9-4575-8839-125d7a50f8bc.azure-devices.net established.[0m
[0;32mI (30559) AZ IOT: Creating an MQTT connection to iotc-c6e37316-0ec9-4575-8839-125d7a50f8bc.azu
re-devices.net.

[0m
[0;32mI (30579) AZ IOT: iotc-c6e37316-0ec9-4575-8839-125d7a50f8bc.azure-devices.net%2Fdevices%2FESPnoPNP1
1650490273[0m
[0;32mI (30679) MQTT: Packet received. ReceivedBytes=2.[0m
[0;32mI (30679) MQTT: CONNACK session present bit set.[0m
[0;32mI (30679) MQTT: Connection accepted.[0m
[0;32mI (30689) MQTT: Received MQTT CONNACK successfully from broker.[0m
[0;32mI (30699) MQTT: MQTT connection established with the broker.[0m
[0;32mI (30699) AZ IOT: An MQTT connection is established with iotc-c6e37316-0ec9-4575-8839-125d7a50f8bc.azure-devices.net[0m
[0;32mI (30789) MQTT: Packet received. ReceivedBytes=3.[0m
[0;32mI (30789) AZ IOT: Suback receive context found: 0x00000001[0m
[0;32mI (30859) MQTT: Packet received. ReceivedBytes=4.[0m
[0;32mI (30859) AZ IOT: Suback receive context found: 0x00000002[0m
[0;32mI (30889) AZ IOT: Successfully sent telemetry message[0m
[0;32mI (30919) AZ IOT: Attempt to receive publish messa
ge from IoT Hub.

[0m
[0;32mI (30969) MQTT: Packet received. ReceivedBytes=176.[0m
[0;32mI (30969) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.[0m
[0;32mI (30969) MQTT: State record updated. New state=MQTTPublishDone.[0m
[0;32mI (30979) AZ IOT: $iothub/twin/res/200/?$rid=2[0m
[0;32mI (30979) AZ IOT: Successfully parsed properties[0m
[0;32mI (30989) AZ IOT: Client updated desired temperature variables locally.[0m
[0;32mI (30989) AZ IOT: Current Temperature: 0.000000[0m
[0;32mI
 (31009) AZ IOT: Maximum Temperature: 22.000000[0m
[0;32mI (31009) AZ IOT: Minimum Temperature: 0.000000[0m
[0;32mI (31009) AZ IOT: Average Temperature: 11.000000[0m
[0;32mI (31129) MQTT: Packet received. ReceivedBytes=2.[0m
[0;32mI (31129) MQTT: Ack packet deserialized with result: MQTTSuccess.[0m
[0;32mI (31129) MQTT: State record updated. New state=MQTTPublishDone.[0m
[0;32mI (31139) AZ IOT: Puback receive
d for packet id: 0x00000003[0m
[0;32mI (31179) MQTT: Packet received. ReceivedBytes=45.[0m
[0;32mI (31179) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.[0m
[0;32mI (31179) MQTT: State record updated. New state=MQTTPu
blishDone.[0m
[0;32mI (31189) AZ IOT: $iothub/twin/res/204/?$rid=5&$version=73904[0m
[0;32mI (31219) MQTT: Packet received. ReceivedBytes=45.[0m
[0;32mI (31219) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.[0m
[0;
32mI (31219) MQTT: State record updated. New state=MQTTPublishDone.[0m
[0;32mI (31229) AZ IOT: $iothub/twin/res/204/?$rid=3&$version=73903[0m
[0;32mI (33239) AZ IOT: Keeping Connection Idle...

[0m
[0;32mI (35269) AZ IOT: Successfully sent telemetry message[0m
[0;32mI (35289) AZ IOT: Attempt to receive publish message from IoT Hub.

[0m
[0;32mI (35469) MQTT: Packet received. ReceivedBytes=2.[0m
[0;32mI (35469) MQTT: Ack packet deserialized with result: MQTTSuccess.[0m
[0;32mI (35469) MQTT: State record updated. New state=MQTTPublishDone.[0m
[0;32mI (35489) AZ IOT: Puback receive
d for packet id: 0x00000004[0m
[0;32mI (35569) MQTT: Packet received. ReceivedBytes=45.[0m
[0;32mI (35569) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.[0m
[0;32mI (35579) MQTT: State record updated. New state=MQTTPublishDone.[0m
[0;32mI (35589) 
AZ IOT: $iothub/twin/res/204/?$rid=7&$version=73905[0m
[0;32mI (37599) AZ IOT: Keeping Connection Idle...

[0m
[0;32mI (39629) AZ IOT: Successfully sent telemetry message[0m
[0;32mI (39649) AZ IOT: Attempt to receive publish message from IoT Hub.

[0m
[0;32mI (39869) MQTT: Packet received. ReceivedBytes=2.[0m
[0;32mI (39869) MQTT: Ack packet deserialized with result: MQTTSuccess.[0m
[0;32mI (39869) MQTT: State record updated. New state=MQTTPublishDone.[0m
[0;32mI (39879) AZ IOT: Puback receive
d for packet id: 0x00000005[0m
[0;32mI (40039) MQTT: Packet received. ReceivedBytes=45.[0m
[0;32mI (40039) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.[0m
[0;32mI (40039) MQTT: State record updated. New state=MQTTPublishDone.[0m

danewalton, 2022-04-21:
Hi @mojy25 
Telemetry can be sent as JSON payloads. There are various ways you could construct those. We have a JSON writer if you wanted to be formulaic about it, or you could make it simpler and use something like an `snprintf()` to create the payloads.

mojy25, 2022-04-21:
Thank you for the reply, 

A simple implementation will do. The application just requires an update periodically, similar to the temperature template created just sending different data.

I've been trying to implement the simple method using snprintf, but I can't seem to have any luck with where this statement should be created.

From my understanding, the VStartDemoTask in azure_iot_freertos_esp32_main.c at line 285 creates an infinite loop, where the device connects to Azure and continues to send data to it via MQTT periodically. 
The sample_azure_iot.c contains all the information for the communication loop, including creating the MQTT message to be sent with telemetry data. 

If that is correct (please correct me if it is not), would I have to set up the message I'd like to send in this sample_azure_iot.c file after the MQTT connection is established, rather than in the azure_iot_freertos_esp32_main.c file? My best guess currently is replacing the Telemetry Bag creation with my data collection and keeping the same template for sending the message there.

Is my logic flawed here?

danewalton, 2022-04-21:
Based on the logs, I believe you are running the Plug and Play sample (I see max, min, and average temperature). We abstract the data collection with [this header](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/sample_azure_iot_pnp/sample_azure_iot_pnp_data_if.h). The base ESP32 sample has no assumptions about whether the device has real sensors, so we have a file which [creates simulated data here](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/sample_azure_iot_pnp/sample_azure_iot_pnp_simulated_data.c). Finally, the function which is specifically called to get the telemetry payload can be found [here](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/59cde927d53bad15f5c4704e7964197497623284/demos/sample_azure_iot_pnp/sample_azure_iot_pnp_simulated_data.c#L471-L489). Feel free to modify that function to create the payload of your choice.

For an `snprintf()` example of how this could be done, please see here: https://godbolt.org/z/o87ccPvh7

mojy25, 2022-04-21:
Thank you, you were right I have the plug and play sample running right now, I thought I had it without the plug and play by deselecting the Using PnP Sample box in the menuconfig for the firmware. 

I can get the ESP32 stuck in the transmission loop now, I'll try getting a message to Azure using what you sent me.

Assuming there's no significant advantage to using the no PnP, I'll try to edit the Plug and Play messages to send what I need

mojy25, 2022-04-22:
I was able to get the Data I needed into the Plug and Play format for proof of concept, I'll have to integrate it all into a clean firmware eventually. Thank you for your help Dane! Much appreciated.

danewalton, 2022-04-22:
Awesome that's great! I'll go ahead and close this for now then. Please feel free to open another issue if you need more help.

closed 2022-04-22 by danewalton