Azure-Samples/iot-middleware-freertos-samples #172: guru meditaion / reboot "AZ IOT: Puback received for packet id" in esp32 sample

dermoritz, 2022-05-16:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> follow the steps from readme here https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/esp32

in my case my CONFIG_IDF_TARGET is "esp32c3"

### Any log messages given by the failure
>
```
I (28275) AZ IOT: Puback received for packet id: 0x00000003
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x4203ecc0  RA      : 0x4203ec7c  SP      : 0x3fca91c0  GP      : 0x3fc8f600
0x4203ecc0: capture_tcp_transport_error at C:/Users/moritz/entwicklung/projekte/bosch_pt/code/esp-idf-v4.4.1/components/tcp_transport/transport.c:332

0x4203ec7c: capture_tcp_transport_error at C:/Users/moritz/entwicklung/projekte/bosch_pt/code/esp-idf-v4.4.1/components/tcp_transport/transport.c:324

TP      : 0x3fc89860  T0      : 0x4005890e  T1      : 0x00000004  T2      : 0x00000080
S0/FP   : 0x00000002  S1      : 0x3fca1e5c  A0      : 0x00000000  A1      : 0x00000002
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fc98000  A5      : 0x00008008
A6      : 0x42024c36  A7      : 0x00000000  S2      : 0x3fca8198  S3      : 0x3fca9244
0x42024c36: lwip_read at C:/Users/moritz/entwicklung/projekte/bosch_pt/code/esp-idf-v4.4.1/components/lwip/lwip/src/api/sockets.c:1313

S4      : 0x00000001  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fcb0564  T4      : 0x3fcb0510  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Failed to run gdb_panic_server.py script: Command '['riscv32-esp-elf-gdb', '--batch', '-n', 'c:\\espbuild\\azure_iot_freertos_esp32.elf', '-ex', 'target remote | "C:\\Users\\moritz\\.espressif\\python_env\\idf4.4_py3.10_env\\Scripts\\python.exe" "C:\\Users\\moritz\\entwicklung\\projekte\\bosch_pt\\code\\esp-idf-v4.4.1\\tools\\idf_monitor_base\\..\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\moritz\\AppData\\Local\\Temp\\tmp4v181ncg"', '-ex', 'bt']' returned non-zero exit status 1.
b'C:\\Users\\moritz\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r2-patch3-8.4.0\\riscv32-esp-elf\\bin\\riscv32-esp-elf-gdb.exe: warning: Couldn\'t determine a path for the index cache directory.\r\nerror starting child process \'"C:\\Users\\moritz\\.espressif\\python_env\\idf4.4_py3.10_env\\Scripts\\python.exe" "C:\\Users\\moritz\\entwicklung\\projekte\\bosch_pt\\code\\esp-idf-v4.4.1\\tools\\idf_monitor_base\\..\\gdb_panic_server.py" --target esp32c3 "C:\\Users\\moritz\\AppData\\Local\\Temp\\tmp4v181ncg"\': CreateProcess: No such file or directory\r\nNo stack.\r\n'

Core  0 register dump:
MEPC    : 0x4203ecc0  RA      : 0x4203ec7c  SP      : 0x3fca91c0  GP      : 0x3fc8f600
TP      : 0x3fc89860  T0      : 0x4005890e  T1      : 0x00000004  T2      : 0x00000080
S0/FP   : 0x00000002  S1      : 0x3fca1e5c  A0      : 0x00000000  A1      : 0x00000002
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3fc98000  A5      : 0x00008008
A6      : 0x42024c36  A7      : 0x00000000  S2      : 0x3fca8198  S3      : 0x3fca9244
S4      : 0x00000001  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fcb0564  T4      : 0x3fcb0510  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fca91c0: 0x3fca8198 0x3fca1e5c 0x00000000 0x4203f566 0x00000000 0x00000000 0x00006b6c 0x3fca930c
3fca91e0: 0x00000001 0x3fca9244 0x3fca930c 0x4203ebba 0x00000001 0x3fc98114 0x3fca9244 0x4200936c
3fca9200: 0x42009312 0x00000000 0x3fca9244 0x4200fb46 0x3c0ba3d8 0x00000002 0x00006b6c 0x0000005a
3fca9220: 0x00000001 0x00000000 0x3fc98114 0x4200df7a 0x3c0b5540 0x3fc93780 0x00000002 0x3fca929c
3fca9240: 0x3c0b5420 0x3c0b5540 0x3fc93780 0x00000002 0x3fc94980 0x00000000 0x3fc98114 0x000001f4
3fca9260: 0x3fc98114 0x00000000 0x0000005a 0x4200e166 0x00000001 0x00000000 0x00000000 0x00000000
3fca9280: 0x00000000 0x000001f4 0x00000000 0x4200c608 0x00000000 0x00000000 0x00000000 0x4200a446
3fca92a0: 0x3c0b55ac 0x00000000 0x00000000 0x42008382 0x420079e2 0x3fca9310 0xa5a5a5a5 0xa5a5a5a5
3fca92c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x0000000d 0x0000001d 0x3fc94ac0 0x3fc94b40 0x01c8f600
3fca92e0: 0x3c0c4fb0 0x00000000 0x3c0b53c4 0x0000001d 0x00000000 0x00000000 0x3c0b6854 0x0000002f
3fca9300: 0x00000000 0x00000000 0x00000000 0x3fca1e5c 0x42009312 0x420093a4 0x3fca930c 0x00000000
3fca9320: 0x00000000 0x3fc8f3e4 0x00000cb9 0x00000000 0x00000000 0x00000000 0x00000000 0x00000015
3fca9340: 0x00000000 0x00000000 0x00000000 0x40389eb8 0x00000000 0x00000000 0x00000000 0x00000000
3fca9360: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0x000000ff
3fca9380: 0xffffffff 0x7fefffff 0x00000000 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3fe00000
3fca93a0: 0x00000000 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3 0x3fc68a28 0x509f79fb 0x3fd34413
3fca93c0: 0x00000000 0x401c0000 0x00000000 0x40140000 0x00000000 0x43500000 0x7fc00000 0x7f7fffff
3fca93e0: 0x00000000 0x40000000 0xffc00000 0x41dfffff 0x94a03595 0x3fdfffff 0x35afe535 0x3fe00000
3fca9400: 0x94a03595 0x3fcfffff 0x00000000 0x7ff80000 0xa5a5a5a5 0xa5a5a5a5 0x00004150 0x00000000
3fca9420: 0x08000000 0x00030317 0x238e4e04 0x7ad7f807 0xc03f7ffb 0x4ea16f2b 0x000000df 0x84a9f000
3fca9440: 0xff81348f 0xe122da3e 0x1d65263f 0xef0f14ba 0xab5d6681 0x894ec4a3 0x643b065f 0x0b0b0bb1
3fca9460: 0x0b0b0b0b 0x0b0b0b0b 0x87a0260b 0x607e3c1e 0x2bf86c57 0x9d8e5c63 0x4177acef 0xdafe3391
3fca9480: 0xa1aa0a50 0x030303d7 0x00000003 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca94a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca94c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca94e0: 0x00000000 0x00000000 0x057c0000 0x423f291a 0x25e97fb2 0xa41faff9 0xab4d4219 0x9daa0771
3fca9500: 0x1e41058d 0x93338948 0x0e0eb939 0x0e0e0e0e 0x0e0e0e0e 0x0e0e0e0e 0x1e45860e 0x9d81361f
3fca9520: 0x047d3cf8 0x70d1e986 0x5bfb6b0f 0xf12da02b 0x549a73b1 0x457d9c6f 0x6de02962 0x1a317277
3fca9540: 0xd2f4e821 0x4b65c13b 0x060403ae 0xed235b52 0x2da3c05d 0x1f6c4376 0xf75ef05d 0xb0920419
3fca9560: 0x14a38e76 0x83aa6cbb 0x63c6dfa9 0x0667e4a3 0x5e8dd6d1 0x000db8a3 0x01031a00 0x12004002
3fca9580: 0x01050104 0x03040102 0x03020305 0x01060202 0x00000306 0x0000000e 0x1eeba2b0 0xb6afeac5
3fca95a0: 0xe3d1be31 0xb7889db5 0xede42c3c 0x066f3bc8 0x116bf1ed 0x815b9371 0x8ceb5c64 0x8f27bbc8

ELF file SHA256: 3190aad83cdd3ed6
ELF file SHA256: 3190aad83cdd3ed6

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380654
0x40380654: esp_restart_noos_dig at C:/Users/moritz/entwicklung/projekte/bosch_pt/code/esp-idf-v4.4.1/components/esp_system/esp_system.c:44 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16b4
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (35) boot: ESP-IDF v4.4.1 2nd stage bootloader
```

### Expected/desired behavior
> This should not happen i guess. After reboot the device will successfully reconnect and send data until next crash.

### OS and Version?
> Windows 11. 

### Versions
> esp-ide 4.4.1 

### Mention any other details that might be useful

(i am new to iot/ embedded world, so this might be a simple mistake of mine ?!)

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

dermoritz, 2022-05-16:
i fixed it temporarily by changing "transport_tls_esp32.c" 👍 

  ``
    //using list due to https://github.com/espressif/esp-idf/issues/8895 causing https://github.com/Azure-Samples/iot-middleware-freertos-samples/issues/172
    esp_transport_list_handle_t mytransportlist = esp_transport_list_init();
    pNetworkContext->xTransport = esp_transport_ssl_init( );
    esp_transport_list_add(mytransportlist, pNetworkContext->xTransport, "_tcp");

danewalton, 2022-05-16:
Interesting thanks for the callout. I wonder if this is the reason our project needed to be pinned to IDF Version 4.3 on Windows. I'll try to look into this.

dermoritz, 2022-05-20:
probably they will fix this soon in esp-idf. the issue i linked seems to be there right from the beginning. It only occurs if error handler is called (but not initialized through list_add).

danewalton, 2022-05-31:
Are you able to try the branch here and see if it helps solve your issue?

https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/dane/transportlist

danewalton, 2022-06-24:
@dermoritz any updates for this?

dermoritz, 2022-06-27:
sorry not yet. we just starting the project (my research was just kind of POC) but it seems we will stick to the samples you are providing. I think we just need a little more time to set up everything - mainly the team. I will send feedback as soon as we checked it.

danewalton, 2022-07-14:
Checking in again here @dermoritz

danewalton, 2022-11-23:
https://github.com/Azure-Samples/iot-middleware-freertos-samples/pull/269 should have fixed this issue. Please let us know if it isn't the case.

closed 2022-11-23 by danewalton