Azure-Samples/iot-middleware-freertos-samples #175: IotHubUnauthorized error with SAS key just after successful DPS (ESP32)

thesparkness, 2022-05-19:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>Hello,

I tried to use the aziotkit to get my ESP connected to Azure Iot Hub. I am running it on a custom board with an ESP32 but the structure is very close to what is on the AZIOTKIT. I am using the Symetric key as seen here: https://youtu.be/PNykfuJ3VDs?t=1567.

I don't understand why the MQTT is unauthorized (noAuth) after the first successful step of the DPS. The ESP crashes on the assert after that.

You can find the log below:

`
00:30:48.328 -> [0;32mI (11509) sample_azureiotkit: Waiting for time synchronization with SNTP server [0m
00:30:52.973 -> W (16139) wifi:<ba-add>idx:0 (ifx:0, 6e:69:eb:7d:a2:e8), tid:0, ssn:3, winSize:64
00:30:53.210 -> [0;32mI (16349) sample_azureiotkit: Notification of a time synchronization event [0m
00:30:53.352 -> [0;32mI (16509) AZ IOT: Creating a TLS connection to [global.azure-devices-provisioning.net](http://global.azure-devices-provisioning.net/):8883.

00:30:53.352 -> [0m
00:30:56.519 -> [0;32mI (19679) tls_freertos: (Network connection 0x3ffc6010) Connection to [global.azure-devices-provisioning.net](http://global.azure-devices-provisioning.net/) established. [0m
00:30:56.519 -> [0;32mI (19679) AZ IOT: 0ne005E2F83%2fregistrations%2ftestingin
00:30:56.519 -> 1652830257 [0m
00:30:56.708 -> [0;32mI (19859) MQTT: Packet received. ReceivedBytes=2. [0m
00:30:56.708 -> [0;32mI (19859) MQTT: CONNACK session present bit not set. [0m
00:30:56.708 -> [0;32mI (19859) MQTT: Connection accepted. [0m
00:30:56.708 -> [0;32mI (19859) MQTT: Received MQTT CONNACK successfully from broker. [0m
00:30:56.708 -> [0;32mI (19879) MQTT: MQTT connection established with the broker. [0m
00:30:56.708 -> [0;32mI (19879) AZ IOT: AzureIoTProvisioning established an MQTT connection with [global.azure-devices-provisioning.net](http://global.azure-devices-provisioning.net/) [0m
00:30:58.768 -> [0;32mI (21899) AZ IOT: AzureIoTProvisioning attempting to subscribe to the MQTT topic: devices/+/messages/devicebound/# [0m
00:30:58.815 -> [0;32mI (21979) MQTT: Packet received. ReceivedBytes=3. [0m
00:31:01.409 -> [0;32mI (24549) MQTT: Packet received. ReceivedBytes=144. [0m
00:31:01.409 -> [0;32mI (24549) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess. [0m
00:31:01.409 -> [0;32mI (24549) MQTT: State record updated. New state=MQTTPublishDone. [0m
00:31:01.409 -> [0;32mI (24559) AZ IOT: $dps/registrations/res/202/?$rid=1&retry-after=3 [0m
00:31:01.409 -> [0;32mI (24579) AZ IOT: {"operationId":"5.18868892921e01xxxxx","status":"assigning"} [0m
00:31:07.972 -> [0;32mI (31119) MQTT: Packet received. ReceivedBytes=466. [0m
00:31:07.972 -> [0;32mI (31119) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess. [0m
00:31:07.972 -> [0;32mI (31119) MQTT: State record updated. New state=MQTTPublishDone. [0m
00:31:07.972 -> [0;32mI (31139) AZ IOT: $dps/registrations/res/200/?$rid=1 [0m
00:31:07.972 -> [0;32mI (31139) AZ IOT: {"operationId":"5.18868892xxxx","status":"assigned","registrationState":{"registrationId":"individual","createdDateTimeUtc":"2022-05-17T22:31:02.127779Z","assignedHub":"[xxxxxx.azure-devices.net](http://xxxxxx.azure-devices.net/)","deviceId":"individual","status":"assigned","substatus":"initialAssignment","lastUpdatedDateTimeUtc":"2022-05-17T22:31:02.3303866Z","etag":"IjVlMDBmNTdhLTAwMDAtMGUwMC0wMDAwxxxxxx="}} [0m
00:31:08.018 -> [0;32mI (31169) AZ IOT: Successfully acquired IoT Hub name and Device ID [0m
00:31:08.018 -> [0;32mI (31189) AZ IOT: Creating a TLS connection to [xxxx.azure-devices.net](http://xxxx.azure-devices.net/):8883.

00:31:08.065 -> [0m
00:31:10.486 -> [0;32mI (33649) tls_freertos: (Network connection 0x3ffc6090) Connection to [xxxxx.azure-devices.net](http://xxxxx.azure-devices.net/) established. [0m
00:31:10.486 -> [0;32mI (33649) AZ IOT: Creating an MQTT connection to [xxxxxx.azure-devices.net](http://xxxxxx.azure-devices.net/).

00:31:10.486 -> [0m
00:31:10.486 -> [0;32mI (33659) AZ IOT: [xxxxxx.azure-devices.net](http://xxxxxx.azure-devices.net/)%2Fdevices%2Findividual%2Fmodules%2Fmainmodule
00:31:10.534 -> 1652830270 [0m
00:31:10.629 -> [0;32mI (33779) MQTT: Packet received. ReceivedBytes=2. [0m
00:31:10.629 -> [0;32mI (33779) MQTT: CONNACK session present bit not set. [0m
00:31:10.629 -> [0;31mE (33779) MQTT: Connection refused: not authorized. [0m
00:31:10.629 -> [0;31mE (33779) MQTT: CONNACK recv failed with status = MQTTServerRefused. [0m
00:31:10.629 -> [0;31mE (33799) MQTT: MQTT connection failed with status = MQTTServerRefused. [0m
00:31:10.676 -> [0;31mE (33799) AZ IOT: Failed to establish MQTT connection: Server=[xxxx.azure-devices.net](http://xxxx.azure-devices.net/), MQTT error=0x00000006 [0m
00:31:10.676 ->
00:31:10.676 -> assert failed: prvAzureDemoTask sample_azure_iot_pnp.c:418 (xResult == eAzureIoTSuccess)`
### Any log messages given by the failure
>
I get this message in my log from Azure IOT
`
{"deviceId":"deviceIDfromDPS","moduleId":"mainmodule","sdkVersion":"c/1.1.0-beta.1(FreeRTOS)","protocol":"Mqtt","authType":null,"maskedIpAddress":"xx.xx.xxx.XXX","statusCode":"401","errorMessage":""}`

### Expected/desired behavior
>The telemetry should start, and I should have the SAS working on my IoT Hub domain (it is public).

### OS and Version?
> Windows  10, ESP IDF 4.3

### Versions
>

### Mention any other details that might be useful
I opened a ticket there: https://docs.microsoft.com/en-us/answers/questions/853365/mqtt-connection-refused-esp-iot-kit-sample-code.html
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

Thanks a lot for your help

danewalton, 2022-05-24:
Hmmm that's interesting. The `"authtype":null` looks to be a bit concerning. Can you confirm on the IoT Hub which is connected to your Device Provisioning Service shows that device as available to connect? So going to the `Devices` section on the Azure Portal, do you see an entry for your device?

danewalton, 2022-05-24:
Actually, can you try connecting the device without a module id?

thesparkness, 2022-05-24:
Hello @danewalton , without the module id it is working perfectly. 
Maybe It is an isolated case but it could be mentioned in the README?
Thanks a lot.

closed 2022-05-24 by thesparkness