Azure-Samples/iot-middleware-freertos-samples #214: [ADU] ESP32 ADU sample freezes when downgrading the image (1.1 to 1.0)

ewertons, 2022-07-15:
> Please provide us with the following information:
> ---------------------------------------------------------------

Device froze after receiving this request:

```json
{"desired":{"deviceUpdate":{"service":{"workflow":{"action":255,"id":"nodeployment"}},"__t":"c"},"$version":4},"reported":{"deviceUpdate":{"__t":"c","agent":{"deviceProperties":{"manufacturer":"ESPRESSIF","model":"ESP32-Azure-IoT-Kit","interfaceId":"dtmi:azure:iot:deviceUpdate;1","aduVer":"DU;agent/0.8.0-rc1-public-preview"},"compatPropertyNames":"manufacturer,model","state":0,"installedUpdateId":"{\"provider\":\"ESPRESSIF\",\"name\":\"ESP32-Azure-IoT-Kit\",\"version\":\"1.1\"}","workflow":{"action":3,"id":"d24626b0-647f-4bba-bac5-d94cad6d48ad-5927a9f1010728cb01c4389b415b34280eb65e5e"}},"service":{"ac":406,"av":3,"value":{}}},"$version":8}}
```

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>

### Any log messages given by the failure
>

### Expected/desired behavior
>

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

alancshi, 2022-07-15:
Repro:

1) I followed steps to create an v1.0 image using ESP-IDF and flashed my device. Before doing anything else, I also created an ADU update for this image.
2) I then created a v1.1 image and also created an ADU update for this.
3) I imported both v1.0 and v1.1 to ADU
4) ADU tells me the latest update for that device is v1.1, as expected, and I deployed that to the device. The device successful reports v1.1 and ADU also says it was successful.
5) Go to ADU and delete v1.1 update
6) ADU will eventually show that the latest update should be v1.0, and let you deploy it. Create a new deployment.

At this point, the device is stuck. ADU reports the update is available, but never says it is in progress. I see no new messages generated on the console for the device. If I look at the device twin, it says the connection state is "disconnected".

Following our in-person debugging, I pressed the reset button on the device. When the device reset, it started to pick up the v1.0 update, and at that point everything proceeded as expected (console reports v1.0 installed, and ADU says it is installed). As far as I can tell, it looks right at this point, and there is a bug requiring me to reboot the device.

FWIW, you guys noticed this in my console log after the device was reset, that seems like it might be a related issue?

>E (195671) AZ IOT: az_iot_adu_client_get_agent_state_payload failed: 0x80010002
>E (195671) AZ IOT: [ADU] Failed sending agent state.

ewertons, 2022-07-15:
I have a strong suspicion it is because the sample is receiving the following in the desired properties:

"desired": {
	"deviceUpdate": {
		"service": {
			"workflow": {
				"action": 255,
				"id": "nodeployment"
			}
		},
		"__t": "c"
	},
	"$version": 4
}

Deleting an update in the portal causes this. It was discussed previously that when an update is deleted in the portal, ADU should not write "nodeployment", it should write the id of the deleted update.

ewertons, 2022-07-15:
But in any case, our code should handle that, so we need to fix it too.

danewalton, 2022-07-15:
The root of the problem comes with this payload being parsed

```json
{
  "service": {
    "workflow": {
      "action": 255,
      "id": "nodeployment"
    }
  },
  "__t": "c"
}
```

And the problematic code is here:

https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/82321fdbc7f26dc38dec4053004505445bd3345a/libs/azure-iot-middleware-freertos/source/azure_iot_adu_client.c#L298-L312

It should not get to line 312 as there is no update manifest.

We should maybe check the action or the id for `nodeployment`.

PR #218 (2022-07-18): "No deployment" scenario for ADU
closes #214
closed 2022-07-19 by danewalton