Azure-Samples/iot-middleware-freertos-samples #224: Kernal Panic Issue even with ESP-IDF V4.3

naakhtar, 2022-07-29:
Kernal Panic Issue with ESP-IDF 

I have tested same Kernal panic issue in V4.3 & V4.4 and below is the outcome.

### Any log messages given by the failure

TELE Timer 0 Val is = 3930
TELE Timer 0 Val is = 3931
I (3932412) fota: Running Fota Stack!
E (3933392) tls_freertos: Reading failed, errno= 128
E (3933392) MQTT: A single byte was not read from the transport: transportStatus=-1.
E (3933392) MQTT: Receiving incoming packet length failed. Status=MQTTRecvFailed
E (3933402) MQTT: Exiting process loop due to failure: ErrorStatus=MQTTRecvFailed
E (3933412) AZ IOT: AzureIoTMQTT_ProcessLoop failed: ProcessLoopDuration=500, MQTT error=0x00000004

assert failed: lum_prvAzureDemoTask lum_app_esp32_provisioning.c:416 (xResult == eAzureIoTSuccess)

Backtrace:0x400816a3:0x3ffcea40 0x40089f95:0x3ffcea60 0x40090e7d:0x3ffcea80 0x400dad19:0x3ffceba0 0x4008d465:0x3ffcec50
0x400816a3: panic_abort at D:/Work/1-Projects/3-ESP-IDF-v4.3.3/Espressif/frameworks/esp-idf-v4.3.3/components/esp_system/panic.c:393

0x40089f95: esp_system_abort at D:/Work/1-Projects/3-ESP-IDF-v4.3.3/Espressif/frameworks/esp-idf-v4.3.3/components/esp_system/system_api.c:112

0x40090e7d: __assert_func at D:/Work/1-Projects/3-ESP-IDF-v4.3.3/Espressif/frameworks/esp-idf-v4.3.3/components/newlib/assert.c:85

0x400dad19: lum_prvAzureDemoTask at D:/Work/1-Projects/3-ESP-IDF-v4.3.3/Espressif/frameworks/esp-idf-v4.3.3/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/aziotkit/components/lum_app/src/lum_app_esp32_provisioning.c:416 (discriminator 1)

0x4008d465: vPortTaskWrapper at D:/Work/1-Projects/3-ESP-IDF-v4.3.3/Espressif/frameworks/esp-idf-v4.3.3/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 9e9b6aa864b62428

Rebooting...
ets Jun 8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7568
load:0x40078000,len:14888
load:0x40080400,len:3688
0x40080400: _init at ??:?

entry 0x4008067c

### OS and Version?
> Windows 10.

danewalton, 2022-07-29:
Hi @naakhtar 

I see this bit in the logs.

```log
assert failed: lum_prvAzureDemoTask lum_app_esp32_provisioning.c:416 (xResult == eAzureIoTSuccess)
```

It seems to be coming from a file called `lum_app_esp32_provisioning.c` which isn't in our repo. Are you able to share the code snippet where this is?

This also just seems like an error in the mbedTLS section which propagates up to a failed assert. Can you give the details of your authentication method? (SAS key or x509 certificate. Group or individual enrollment).

naakhtar, 2022-07-29:
SAS key individual enrollment.

x509 certificate group enrollment procedure not clear. Need help on that how to do that.

danewalton, 2022-08-01:
The error is coming from a file which is not ours. Are you able to share the file that you are using?

The panic is due to an assert that is failing.

danewalton, 2022-11-23:
Closing this issue for now due to inactivity. If you would like it reopened, please let us know. For any other questions, please open another Github issue.

closed 2022-11-23 by danewalton