Azure-Samples/iot-middleware-freertos-samples #225: Issue with Group enrollment for X.509 certificate

naakhtar, 2022-07-29:
Not able to do provisioning Service enrollment group for X.509 certificate attestation.

ramanijaldeep, 2022-09-19:
Not able to provision the device with enrollment group service for X.509 certificate attestation.

danewalton, 2022-09-19:
Hi @ramanijaldeep 

1. How are you configuring your certificates? What commands are you using?
2. How are you adding those certificates to your demo config? What format are they in?

ramanijaldeep, 2022-10-10:
> Hi @ramanijaldeep
> 
> 1. How are you configuring your certificates? What commands are you using?
> 2. How are you adding those certificates to your demo config? What format are they in?

Hi @danewalton 

In sample examples, it is working. I got an error when building an application on FreeRTOS devices.

danewalton, 2022-10-10:
Sorry your answer is a little confusing to me. Does this mean you don't need help anymore and you have your solution working?

danewalton, 2023-01-09:
Closing this issue for now due to inactivity. If you would like it reopened, please let us know. For any other questions, please open another Github issue.

closed 2023-01-09 by danewalton