Azure-Samples/iot-middleware-freertos-samples #254: ADU ESPRESSIF ESP-IDF v4.4 Failed to open a new connection: 258

jochemdevos, 2022-10-24:
### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Working with branch ota (/Azure-Samples/iot-middleware-freertos-samples/tree/ota/demos/projects/ESPRESSIF/adu)
> Build with ESP-IDF v4.4, Create manifest and upload to storage account
> Create/Update device update deployment with these files
> Desired device twin is updated
> Monitor device: device receives device twin update, tries to update, but is unable to connect to ADU instance endpoint to fetch files

### Any log messages given by the failure
I (13790) AZ IOT: [ADU] Send property update.
I (13820) AZ IOT: [ADU] Invoke HTTP Connect Callback.
I (13820) AZ IOT: Connecting socket to wb24g01-d-adui--wb24g01-d-adu.b.nlu.dl.adu.microsoft.com
E (13820) TRANSPORT_BASE: Failed to open a new connection: 258
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

### Expected/desired behavior
> Correctly update device, as it does when building with ESP-IDF v4.3

### OS and Version?
> Windows 11

### Versions
> ESP-IDF v4.3
> ota branch

### Mention any other details that might be useful
>

danewalton, 2022-10-24:
Hi @jochemdevos 
Can you please try the `feature/iot-adu` branch? That is our current working branch for the upcoming ota feature. `ota` is an old one. I'll look to get that one removed.

jochemdevos, 2022-10-25:
checked out feature/iot-adu, Installed esp-idf 4.4.2 and a new tools directory, started build resulting in
`CMake Error at C:/Users/jocvos/esp/4.4.2/esp-idf/tools/cmake/component.cmake:300 (message):
  Include directory
  'C:/Github/iot-middleware-freertos-samples/libs/azure-iot-middleware-freertos/libraries/coreMQTT/source/include'
  is not a directory.
Call Stack (most recent call first):
  C:/Users/jocvos/esp/4.4.2/esp-idf/tools/cmake/component.cmake:481 (__component_add_include_dirs)
  components/coreMQTT/CMakeLists.txt:30 (idf_component_register)`

Noticed there was a submodule link pointing to iot-middleware-freertos-samples/libs/azure-iot-middleware-freertos where i had no files locally. Turns out when checking out a git repository submodules are not included by default, so be sure to recursively init submodules.

git submodule update --init --recursive
git submodule update --recursive

https://stackoverflow.com/questions/10168449/git-update-submodules-recursively

After succesfully building and deploying the image via an ADU update, it (connects succesfully to ADU due to the previous image being built with ESP-IDF 4.3), it boots with the new image built with ESP-IDF 4.4. The log then outputs unable to connect to ADU and gets stuck in a bootloop:

I (26439) AZ IOT: Connecting socket to wb24g01-d-adui--wb24g01-d-adu.b.nlu.dl.adu.microsoft.com
E (26439) TRANSPORT_BASE: Failed to open a new connection: 258
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

More complete logging:
```
I (91689) AZ IOT: ------------------------------------------------------------------------------
I (91689) AZ IOT: ADU SAMPLE
I (91689) AZ IOT: Version: 1.10
I (91699) AZ IOT: ------------------------------------------------------------------------------
I (91699) AZ IOT: Creating a TLS connection to wb24g01-tse-d-iothub.azure-devices.net:8883.
I (93199) tls_freertos: (Network connection 0x3ffcee40) Connection to wb24g01-tse-d-iothub.azure-devices.net established.
I (93199) AZ IOT: Creating an MQTT connection to wb24g01-tse-d-iothub.azure-devices.net.
I (93209) AZ IOT: wb24g01-tse-d-iothub.azure-devices.net%2Fdevices%2FJochem_Test_4G
1666693330
I (93309) MQTT: Packet received. ReceivedBytes=2.
I (93309) MQTT: CONNACK session present bit set.
I (93309) MQTT: Connection accepted.
I (93319) MQTT: Received MQTT CONNACK successfully from broker.
I (93329) MQTT: MQTT connection established with the broker.
I (93329) AZ IOT: An MQTT connection is established with wb24g01-tse-d-iothub.azure-devices.net
I (93389) MQTT: Packet received. ReceivedBytes=3.
I (93389) AZ IOT: Suback receive context found: 0x00000001
I (93459) MQTT: Packet received. ReceivedBytes=4.
I (93459) AZ IOT: Suback receive context found: 0x00000002
I (93489) AZ IOT: Successfully sent telemetry message
I (93489) AZ IOT: Attempt to receive publish message from IoT Hub.
I (94119) MQTT: Packet received. ReceivedBytes=4223.
I (94129) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
I (94129) MQTT: State record updated. New state=MQTTPublishDone.
I (94139) AZ IOT: $iothub/twin/res/200/?$rid=2
I (94139) AZ IOT: Writable properties received: {"desired":{"deviceUpdate":{"service":{"workflow":{"action":3,"id":"890aac7b-f454-4b33-b2b6-e2cc861b2f3e"},"updateManifest":"{\"manifestVersion\":\"4\",\"updateId\":{\"provider\":\"ESPRESSIF\",\"name\":\"ESP32-Azure-IoT-Kit\",\"version\":\"1.10\"},\"compatibility\":[{\"deviceManufacturer\":\"ESPRESSIF\",\"deviceModel\":\"ESP32-Azure-IoT-Kit\"}],\"instructions\":{\"steps\":[{\"handler\":\"microsoft/swupdate:1\",\"files\":[\"f038628483159e7aa\"]}]},\"files\":{\"f038628483159e7aa\":{\"fileName\":\"azure_iot_freertos_esp32-v1.10.bin\",\"sizeInBytes\":868656,\"hashes\":{\"sha256\":\"blBsleJXPDEUNcRmeiq5iIistHI4IhhG02QXNTEzgdU=\"}}},\"createdDateTime\":\"2022-10-25T09:04:42.7019915Z\"}","updateManifestSignature":"eyJhbGciOiJSUzI1NiIsInNqd2siOiJleUpoYkdjaU9pSlNVekkxTmlJc0ltdHBaQ0k2SWtGRVZTNHlNREEzTURJdVVpSjkuZXlKcmRIa2lPaUpTVTBFaUxDSnVJam9pYkV4bWMwdHZPRmwwWW1Oak1sRXpUalV3VlhSTVNXWlhVVXhXVTBGRlltTm9LMFl2WTJVM1V6Rlpja3BvV0U5VGNucFRaa051VEhCVmFYRlFWSGMwZWxndmRHbEJja0ZGZFhrM1JFRmxWVzVGU0VWamVEZE9hM2QzZVRVdk9IcExaV3AyWTBWWWNFRktMMlV6UWt0SE5FVTBiMjVtU0ZGRmNFOXplSGRQUzBWbFJ6QkhkamwzVjB3emVsUmpUblprUzFoUFJGaEdNMVZRWlVveGIwZGlVRkZ0Y3pKNmJVTktlRUppZEZOSldVbDBiWFpwWTNneVpXdGtWbnBYUm5jdmRrdFVUblZMYXpob2NVczNTRkptYWs5VlMzVkxXSGxqSzNsSVVVa3dZVVpDY2pKNmEyc3plR2d4ZEVWUFN6azRWMHBtZUdKamFsQnpSRTgyWjNwWmVtdFlla05OZW1Fd1R6QkhhV0pDWjB4QlZGUTVUV1k0V1ZCd1dVY3lhblpQWVVSVmIwTlJiakpWWTFWU1RtUnNPR2hLWW5scWJscHZNa3B5SzFVNE5IbDFjVTlyTjBZMFdubFRiMEoyTkdKWVNrZ3lXbEpTV2tab0wzVlRiSE5XT1hkU2JWbG9XWEoyT1RGRVdtbHhhemhJVWpaRVUyeHVabTVsZFRJNFJsUm9SVzF0YjNOVlRUTnJNbGxNYzBKak5FSnZkWEIwTTNsaFNEaFpia3BVTnpSMU16TjFlakU1TDAxNlZIVnFTMmMzVkdGcE1USXJXR0owYmxwRU9XcFVSMkY1U25Sc2FFWmxWeXRJUXpVM1FYUkJSbHBvY1ZsM2VVZHJXQ3M0TTBGaFVGaGFOR0V4VHpoMU1qTk9WVWQxTWtGd04yOU5NVTR3ZVVKS0swbHNUM29pTENKbElqb2lRVkZCUWlJc0ltRnNaeUk2SWxKVE1qVTJJaXdpYTJsa0lqb2lRVVJWTGpJeE1EWXdPUzVTTGxNaWZRLlJLS2VBZE02dGFjdWZpSVU3eTV2S3dsNFpQLURMNnEteHlrTndEdkljZFpIaTBIa2RIZ1V2WnoyZzZCTmpLS21WTU92dXp6TjhEczhybXo1dnMwT1RJN2tYUG1YeDZFLUYyUXVoUXNxT3J5LS1aN2J3TW5LYTNkZk1sbkthWU9PdURtV252RWMyR0hWdVVTSzREbmw0TE9vTTQxOVlMNThWTDAtSEthU18xYmNOUDhXYjVZR08xZXh1RmpiVGtIZkNIU0duVThJeUFjczlGTjhUT3JETHZpVEtwcWtvM3RiSUwxZE1TN3NhLWJkZExUVWp6TnVLTmFpNnpIWTdSanZGbjhjUDN6R2xjQnN1aVQ0XzVVaDZ0M05rZW1UdV9tZjdtZUFLLTBTMTAzMFpSNnNTR281azgtTE1sX0ZaUmh4djNFZFNtR2RBUTNlMDVMRzNnVVAyNzhT [... cut ...]
0x400d800f: prvConnectHTTP at C:/Github/iot-middleware-freertos-samples/demos/sample_azure_iot_adu/sample_azure_iot_adu.c:373

0x400d80f7: prvDownloadUpdateImageIntoFlash at C:/Github/iot-middleware-freertos-samples/demos/sample_azure_iot_adu/sample_azure_iot_adu.c:471

0x400d8794: prvAzureDemoTask at C:/Github/iot-middleware-freertos-samples/demos/sample_azure_iot_adu/sample_azure_iot_adu.c:876

0x4008b789: vPortTaskWrapper at C:/Users/jocvos/esp/4.4.2/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 26f4020a91c6bdfe

Rebooting...
```

danewalton, 2022-10-25:
I wonder if this is related to this issue: https://github.com/Azure-Samples/iot-middleware-freertos-samples/issues/256

Looking into it.

jochemdevos, 2022-10-31:
hi @danewalton , any update on this particular issue? Kr, Jochem

danewalton, 2022-10-31:
I think I have a fix. Should hopefully have it in PR shortly.

jochemdevos, 2022-11-08:
hi @danewalton, good to hear. Any chance to see it merged soon?

danewalton, 2022-11-10:
Hi @jochemdevos 
can you try this branch here and see if it helps resolve your issue?
https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/dane/transportfix

jochemdevos, 2022-11-14:
different issue with building with dane/transportfix, even after doing 'full clean':

FAILED: esp-idf/sample-azure-iot/CMakeFiles/__idf_sample-azure-iot.dir/63428885650a849b33ae59e107173359/demos/sample_azure_iot_adu/sample_azure_iot_pnp_simulated_data.c.obj 
...
C:/Github/iot-middleware-freertos-samples/demos/sample_azure_iot_adu/sample_azure_iot_pnp_simulated_data.c:148:8: error: unknown type name 'AzureIoTJWS_RootKey_t'
 static AzureIoTJWS_RootKey_t xADURootKeys[] =

full output:
```[1/11] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Github\iot-middleware-freertos-samples\demos\projects\ESPRESSIF\adu\build\bootloader\esp-idf\esptool_py && C:\Users\jocvos\esptools\4.4.2\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/jocvos/esp/4.4.2/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Github/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/adu/build/bootloader/bootloader.bin"
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
[2/9] Building C object esp-idf/sample-azure-iot/CMakeFiles/_...ample_azure_iot_adu/sample_azure_iot_pnp_simulated_data.c.obj
FAILED: esp-idf/sample-azure-iot/CMakeFiles/__idf_sample-azure-iot.dir/63428885650a849b33ae59e107173359/demos/sample_azure_iot_adu/sample_azure_iot_pnp_simulated_data.c.obj 
C:\Users\jocvos\esptools\4.4.2\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -IC:/Github/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/adu/build/config -IC:/Github/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/adu/config -IC:/Github/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/adu/components/sample-azure-iot -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/mbedtls/mbedtls/include -IC:/Github/iot-middleware-freertos-samples/demos/common/transport -IC:/Github/iot-middleware-freertos-samples/demos/common/utilities -IC:/Github/iot-middleware-freertos-samples/demos/sample_azure_iot_adu -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/newlib/platform_include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/freertos/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/freertos/include/esp_additions/freertos -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/freertos/port/xtensa/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/freertos/include/esp_additions -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_hw_support/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_hw_support/include/soc -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_hw_support/include/soc/esp32 -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_hw_support/port/esp32/. -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_hw_support/port/esp32/private_include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/heap/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/log/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/lwip/include/apps -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/lwip/include/apps/sntp -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/lwip/lwip/src/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/lwip/port/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/soc/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/soc/esp32/. -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/soc/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/hal/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/hal/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/hal/platform_port/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_rom/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_rom/include/esp32 -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_rom/esp32 -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_common/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_system/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_system/port/soc -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_system/port/public_compat -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/xtensa/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/xtensa/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/driver/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/driver/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_pm/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_ringbuf/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/efuse/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/efuse/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/vfs/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_wifi/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_event/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_netif/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_eth/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/tcpip_adapter/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_phy/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_phy/esp32/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_ipc/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/app_trace/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/esp_timer/include -IC:/Users/jocvos/esp/4.4.2/esp-idf/components/mbedtls/port/include - [... cut ...]
                 from C:/Github/iot-middleware-freertos-samples/demos/sample_azure_iot_adu/sample_azure_iot_pnp_simulated_data.c:4:
C:/Github/iot-middleware-freertos-samples/demos/projects/ESPRESSIF/adu/config/demo_config.h:135:22: warning: 'root_cert_array' defined but not used [-Wunused-variable]
 static unsigned char root_cert_array[] = {
                      ^~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.

 *  The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command ninja " terminated with exit code: 1. ```

jochemdevos, 2022-11-14:
Hi Dane, that indeed helped resolve the building issue. 

Now running into a different issue with ADU. Since we have deleted and recreated it (since west europe is now available) it seems it can no longer fetch device groups and i therefore cannot deploy an update.

I will add this to my support request, but should you have any info on this please let me know .

danewalton, 2022-11-14:
You should be able to now create a device and put in a tag for the device group. Once the device connects, the ADU service should automatically create a group based on the tag. Let me know if there's another way I can help.

jochemdevos, 2022-11-15:
Hi Dane, our IoThub was unchanged and still contained the previously used devices. Additionally ive created a new one to see if that would trigger something, but no luck.

[image]

[image]

jochemdevos, 2022-11-18:
Hey @RBartonMesh , thanks for the input. unfortunately we've also added the permissions to the new ADU resource so that doesn't seem to be the issue. 

Most likely the ADU service got 'confused' because we have removed (N-EU) and recreated ADU (W-EU, new region since GA) without changing any of the iothub devices. Ive a support request running for this issue.

Once that's resolved I hope to finally retest with branch dane/transportfix :)

Kr, Jochem

KZeronimo, 2022-11-18:
Hi @danewalton, you mentioned that if your transportfix br didn't work that this might be an issue with ADU on the service side is that correct?

danewalton, 2022-11-18:
No if the device is panicking then that's a device issue.

Just to be clear, if you're running esp idf `v4.4.2`, we don't yet have support for that. The way the transport is created is different. I can look into this but you're running an unsupported version.

danewalton, 2022-11-18:
I think I might know what's going on though. The panic looks to be happening when it's accessing the `err`. And in `v4.4.2`, the err variable is created in the `list_init()`, which we don't do yet.

[image]

If you want to try a quick-and-dirty possible fix, I put the changes from a preliminary list addition in with the branch I sent you and put it here: https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/dane/transportfixwithlist

You can try that and see if it helps in your scenario.

cal-kulator, 2022-11-18:
Thank you @danewalton you were correct, we were using an unsupported version of esp-idf. Building on v4.3.4 fixes the error.

KZeronimo, 2022-11-18:
@cal-kulator I suppose Dane's branch transportfixwithlist would work with 4.4.2

Our fun will be actually stepping back to ESP-IDF 4.2.1 to align with the requirements of our project's requirements :)

Thanks @danewalton  for your help!

danewalton, 2022-11-19:
Thanks for letting me know. I think that network list idea is what has been preventing us from moving to 4.4 ([another issue was opened here but we never got confirmation](https://github.com/Azure-Samples/iot-middleware-freertos-samples/issues/172)). 

I'll look to get this merged for us so we can move to support 4.4.

jochemdevos, 2022-11-21:
Fixed my ADU issue (was indeed RBAC for ADU microsoft service, not the ADU resources managed identity as I mistakenly assumed)

with dane/transportfix I still get the 258.

with dane/transportfixwithlist it seems to work well/better! It is able to fetch a new version and boot with it. It does seem however that it keeps fetching the latest update on boot, even if it is already running it.. Would that be a side effect of the fixes in this branch? Or did I stumble on a new issue (perhaps with the service itself)

danewalton, 2022-11-21:
That could be a side effect of how you're creating your update. The device will check if the installed criteria matches what is in the update:

https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/sample_azure_iot_adu/sample_azure_iot_pnp_simulated_data.c#L337-L357

[On our docs from here](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/projects/ESPRESSIF/adu/README.md#generate-the-adu-update-manifest), it would be this value:

[image]

Make sure that matches whatever is listed here: https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/7713ef85946fe6561450207b8b41782740769e68/demos/projects/ESPRESSIF/adu/config/demo_config.h#L489

danewalton, 2022-11-23:
https://github.com/Azure-Samples/iot-middleware-freertos-samples/pull/269 should have fixed this issue. Please let us know if it isn't the case.