Azure-Samples/iot-middleware-freertos-samples #256: Demo project for ESP32 use wrong NetworkContext

matt-salvo, 2022-10-25:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>Custom project based on demo code from 
https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/ota/demos/sample_azure_iot/sample_azure_iot.c
https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/ota/demos/projects/ESPRESSIF/esp32/main/azure_iot_freertos_esp32_main.c

### Any log messages given by the failure
> Guru Meditation Error: Core  0 panic'ed (Store access fault).
> 0x4209e8fc in capture_tcp_transport_error (t=t@entry=0x3fcb01c4, error=error@entry=ERR_TCP_TRANSPORT_CONNECTION_CLOSED_BY_FIN) at /home/user/esp-idf-v4.4.1/components/tcp_transport/transport.c:332
>332                 err_handle->last_error = ESP_ERR_ESP_TLS_TCP_CLOSED_FIN;

### Expected/desired behavior
>err_handle shouldn't be NULL pointer  

### OS and Version?
> Linux Kubuntu 20.04

### Versions
> 

### Mention any other details that might be useful

> ---------------------------------------------------------------
> ESP32 example implement transport layer glue logic whit API in  
  https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/ota/demos/projects/ESPRESSIF/esp32/components/sample-azure-iot/transport_tls_esp32.c 
This module define the network context 
```
    struct NetworkContext
    {
        esp_transport_handle_t xTransport;
        uint32_t ulReceiveTimeoutMs;
        uint32_t ulSendTimeoutMs;
    };
```
to be passed to TLS_Socket_Connect() 

In sample_azure_iot.c the API TLS_Socket_Connect() is called with a different network context
```
    struct NetworkContext
    {
        TlsTransportParams_t * pParams;
    };
```
The code initially seems to work, but this lend to a TCP disconnect followed by the failure reported.

danewalton, 2022-10-25:
Hi @matt-salvo can you please use the `feature/iot-adu` branch and not `ota`. `ota` was a working branch and the feature branch is the one we are using now. I'll look to delete the `ota` one to prevent confusion.

matt-salvo, 2022-10-25:
Hi @danewalton, I corrected the links. I meant to refer to the main branch.

danewalton, 2022-10-25:
Gotcha thanks for the note. Wonder why we haven't hit this issue ourselves. I'll look into this.

closed 2022-11-23 by danewalton

matt-salvo, 2022-11-26:
I just wanted to point out the inconsistency. I solved myself the issue using the correct struct NetworkContext; but I still wonder why is not used esp_tls component which is officially documented by Espressif.

danewalton, 2022-11-28:
@matt-salvo 
just to double check here, do you think our solution is not fully correct?