Azure-Samples/iot-middleware-freertos-samples #285: utf8 encoding and custom attributes are needed for routing queries

devotip, 2022-11-29:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [X] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>build and run an example sending telemetry

### Any log messages given by the failure
>no

### Expected/desired behavior
>data received in iothub is utf-8, not just a byte array and/or there are custom attributes, both are needed to apply a routing query

### OS and Version?
> build environment is windows10, target device is Espressif ESP32 framework esp-idf 4.4.2

### Versions
>

### Mention any other details that might be useful
Custom attributes are not in all examples, while the payload is utf8 encoded is not declared as such making it unusable in the routing filter
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

ewertons, 2023-02-16:
We are adding an example of how to set content type and encoding in the samples:
https://github.com/Azure-Samples/iot-middleware-freertos-samples/pull/337
Please refer to that for the usage.
Thanks,
Azure IoT SDK Team.

ewertons, 2023-02-16:
Thanks. Please feel free to reopen the issue if you would like to follow up.

closed 2023-02-16 by ewertons