Azure-Samples/iot-middleware-freertos-samples #298: core error=0x80010002 establishing TLS session with IoT Hub

afcec, 2022-12-20:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [X ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
I am moving from ESP-AZURE to Azure IoT middleware for Freertos and using a custom board based on ESP32-WROOM-32D. My client certificates are stored in a specific partition. I am using sample_azure_iot.c as an example for my migration and so far I am able to connect to DPS and obtain successfully the IoT Hub name and device ID. However, when I then make the call to:

xResult = prvConnectToServerWithBackoffRetries( ( const char * ) ioTHubHostName,IOTHUB_PORT,&xNetworkCredentials, &xNetworkContext);

I obtain xResult = 4 and I can see the following message:  **AZ IOT: Failed to get username: core error=0x80010002**

Any suggestion?

### Any log messages given by the failure
>
 AZ IOT: Failed to get username: core error=0x80010002
### Expected/desired behavior
>

### OS and Version?
> Windows 10 and ESP-IDF 4.4.3

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

danewalton, 2022-12-20:
Hi @afcec 
The middlware for FreeRTOS is built on top of our [Embedded C SDK](https://github.com/Azure/azure-sdk-for-c).

That error has a "facility code" of 1 (0x800**1**0002) meaning a core error, with the core error being a 2. That leads to this value here [AZ_ERROR_NOT_ENOUGH_SPACE ](https://github.com/Azure/azure-sdk-for-c/blob/59e7637153db5d4dbe423c1acd5cf3b449cc57f9/sdk/inc/azure/core/az_result.h#L77). It seems the buffer into which you are trying to put the username is not big enough. Can you try making it bigger and see if that helps?

afcec, 2022-12-21:
Hi @danewalton , thanks for your help. You were right, I was passing a wrong value as module ID length. 

Now I am receving a MQTT error but I need to dig a little deeper:

** The client certificate is correct as I am able to use it as it is with my previous FW version. The deviceId received as a result of the call to 'AzureIoTProvisioningClient_GetDeviceAndHub' is also correct.

This is the output I get:

COMMUNICATION_BOARD: Initializing hub client with deviceId: 1002202100000168 and deviceId length: 16
E (83831) MQTT: Connection refused: not authorized.
E (83831) MQTT: CONNACK recv failed with status = MQTTServerRefused.
E (83851) MQTT: MQTT connection failed with status = MQTTServerRefused.
E (83941) AZ IOT: Failed to establish MQTT connection: Server=fakeiothubname.azure-devices.net, MQTT error=0x00000006

danewalton, 2022-12-22:
What kind of cert are you using for your device? Is it CA signed or self signed?

afcec, 2022-12-23:
I am using self signed certificates

afcec, 2023-01-05:
Hi, I have not been able to make any progress with this "MQTT connection refused" issue. Any suggestion?

The only answer I can come up with is that in my previous FW version using ESP-AZURE, the device certificate was PEM formatted, that is, each line finishing with \r\n but now according to your documentation [here](https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/esp32) y removed all the carriage returns and left only the line feeds.

danewalton, 2023-01-05:
Are you able to confirm that the Common Name for the certificate matches the device id and registration ID?

If you would like to, you can send me your connection info like dps registration id, id scope, etc and the device cert to my email which I have in my Github profile so that I can look further into the scenario.

afcec, 2023-01-10:
Just some more information, I also tried to use this certificate with the sample "iot-middleware-freertos-samples\demos\projects\ESPRESSIF\esp32" and I receive the same error:

AZ IOT: Successfully acquired IoT Hub name and Device ID
I (55084) AZ IOT: Creating a TLS connection to iot-hub-teka-smart-home.azure-devices.net:8883.

I (62014) tls_freertos: (Network connection 0x3ffcab50) Connection to XXXXXX.azure-devices.net established.
I (62014) AZ IOT: Creating an MQTT connection to XXXXXXX.azure-devices.net.

I (62194) MQTT: Packet received. ReceivedBytes=2.
I (62194) MQTT: CONNACK session present bit not set.
E (62194) MQTT: Connection refused: not authorized.
E (62194) MQTT: CONNACK recv failed with status = MQTTServerRefused.
E (62204) MQTT: MQTT connection failed with status = MQTTServerRefused.
E (62214) AZ IOT: Failed to establish MQTT connection: Server=XXXXXX.azure-devices.net, MQTT error=0x00000006

danewalton, 2023-01-10:
Got your email thanks for sharing the info. I will try and recreate the scenario.

afcec, 2023-01-26:
For those experiencing the same issue, I just needed to remove the module ID from the config. In my case just removing these two lines before the call to AzureIoTHubClient_Init:

xHubOptions.pucModuleID = iotHubDeviceId;
xHubOptions.ulModuleIDLength = strlen((const char *) iotHubDeviceId);

danewalton, 2023-01-27:
Yes you should not be adding a module id if you don't need one. **NOTE** that a `module id != device id`.

For more info on module id and its utility, please see here: https://learn.microsoft.com/en-us/azure/iot-hub/iot-hub-csharp-csharp-module-twin-getstarted

I will close this for now since it seems this problem is resolved.

closed 2023-01-27 by danewalton