Azure-Samples/iot-middleware-freertos-samples #299: Digicert G2 esp-tls-mbedtls: verification info:   ! The certificate is not correctly signed by the trusted CA

hauserkristof, 2022-12-21:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>
- ESP-IDF 4.3.4
- IoTHub is switched to Digicert G2 from Baltimore with [migration tool](https://learn.microsoft.com/en-us/azure/iot-hub/migrate-tls-certificate?tabs=portal)
- Using version v1.0.0 of azure-iot-middleware-freertos (commit id: ```76c74cea9e4fb13a411e41192e332f340309362f ```)
- Using sample with esp-cryptoauthlib
### Any log messages given by the failure
>

```bash
I (17489) transport: esp_transport_ssl_use_secure_element is true
I (17529) esp-tls-mbedtls: Initialize the ATECC interface...
```

<details>

<summary>MbedTLS debug:</summary>

```bash
I (17739) mbedtls: ssl_tls.c:5904 => handshake

I (17739) mbedtls: ssl_cli.c:4483 client state: 0

I (17739) mbedtls: ssl_msg.c:2102 => flush output

I (17749) mbedtls: ssl_msg.c:2114 <= flush output

I (17749) mbedtls: ssl_cli.c:4483 client state: 1

I (17759) mbedtls: ssl_msg.c:2102 => flush output

I (17759) mbedtls: ssl_msg.c:2114 <= flush output

I (17769) mbedtls: ssl_cli.c:999 => write client hello

I (17779) mbedtls: ssl_msg.c:2542 => write handshake message

I (17779) mbedtls: ssl_msg.c:2701 => write record

I (17789) mbedtls: ssl_msg.c:2102 => flush output

I (17789) mbedtls: ssl_msg.c:2122 message length: 256, out_left: 256

I (17799) mbedtls: ssl_msg.c:2127 ssl->f_send() returned 256 (-0xffffff00)

I (17809) mbedtls: ssl_msg.c:2155 <= flush output

I (17809) mbedtls: ssl_msg.c:2870 <= write record

I (17819) mbedtls: ssl_msg.c:2678 <= write handshake message

I (17819) mbedtls: ssl_cli.c:1467 <= write client hello

I (17829) mbedtls: ssl_cli.c:4483 client state: 2

I (17839) mbedtls: ssl_msg.c:2102 => flush output

I (17839) mbedtls: ssl_msg.c:2114 <= flush output

[... 812 lines cut ...]

        /* Attempt to establish TLS session with IoT Hub. If connection fails,
         * retry after a timeout. Timeout value will be exponentially increased
         * until  the maximum number of attempts are reached or the maximum timeout
         * value is reached. The function returns a failure status if the TCP
         * connection cannot be established to the IoT Hub after the configured
         * number of attempts. */
        ulStatus = prvConnectToServerWithBackoffRetries( ( const char * ) pucIotHubHostname,
                                                         configIOTHUB_PORT,
                                                         &xNetworkCredentials, &xNetworkContext );
        configASSERT( ulStatus == 0 );

        /* Fill in Transport Interface send and receive function pointers. */
        xTransport.pxNetworkContext = &xNetworkContext;
        xTransport.xSend = TLS_Socket_Send;
        xTransport.xRecv = TLS_Socket_Recv;
...

```

@danewalton do you have any idea what could be the issue?

If you need any more details, I can provide it of course.

Thank you very much!

danewalton, 2022-12-22:
Thanks @hauserkristof 

Just out of curiosity, is there any way for you to test our sample without using the ATECC? Even a device created with hardcoded SAS etc for simplicity. I want to see if this is mainly in our ATECC or all our samples.

hauserkristof, 2022-12-22:
Hi @danewalton 

Yeah, I'm trying with SAS auth, but my firmware is a bit more hardwired to ATECC, that I have previously thought, if I get any results, I'll inform you.

hauserkristof, 2022-12-22:
Well @danewalton 

I have disabled ATECC (with DPS) and enabled SAS, with DPS successfully.

I can connect OK to our production IoTHub that uses Baltimore cert.

But still failing on the staging IoTHub that uses Digicert.

I saw that you merged a PR #[Add CA verify error to TLS](https://github.com/Azure-Samples/iot-middleware-freertos-samples/pull/293#top) #293 yesterday, which may can provide additional info. Unfortunately it would require me far more time to implement this too.

Until then, you did not have any problem like mine? Could you try it out maybe?

I may try with the exact sample uploaded, than I can see if I have messed something up.

Thanks

hauserkristof, 2022-12-23:
Hi @danewalton

I've tried a fresh sample (espressiff/adu) with SAS, no DPS, the issue is also present there, if I try to connect to an IoTHub with the Digicert G2 cert. Baltimore cert is fine of course.

I can send you both certificate chains, if it helps with anything (Github does not allow .cer uploads). 

But:
Baltimore chain:
[image]

Digicert chain:
[image]

I have added the ```MST RS256 CA-1``` intermediate cert manually, and the issue is still present.

wreyford, 2022-12-30:
I face a similar problem. Updated to Digicert G2 for my IoTHub. **I'm using the Sample from this repo**, but updated to work with ESP-IDF5.1, but only changed as per the 5.0 changes recently pushed. I understand that the usecase is much more advanced above, but feel it may contribute to the solution.

I compile the with `Use PnP in Azure Sample` selected, as well as `Enable Device Provisioning Sample` selected in KConfig.
If I comment out the Baltimore certificate and compile, the DPS enrollment works for an individual enrollment.
It then fails with the same error when connecting to the iot hub for telemetry.
`esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700`
I`esp-tls-mbedtls: Failed to verify peer certificate!`

If I uncomment the Baltimore Certificate, then the DPS fails, but the IOT Hub telemetry works.

So: I need to provision my device with the Baltimore commented out, with with `Use PnP in Azure Sample` un-selected.
Then I need to uncomment the Baltimore certificate, unselect `Enable Device Provisioning Sample`, with  `Enable Device Provisioning Sample` selected, to get PnP to work.

So somehow it appears that when Baltimore is in the chain of three certificates as in the Sample, and you have migrated to G2, then DPS won't work, but  PnP will work. DPS will only work with Baltimore commented out, but then PnP wont work 
So Azure Device Provisioning Service appears to only want the DigiCert Global Root G2, and Microsoft RSA Root Certificate Authority 2017 in the root_cert_array in demo_config.h after migration to G2.
Azure IoT Hub appears to want all three certificates in root_cert_array in demo_config.h after migration to G2.

Hope this can shed some light on the root cause.

danewalton, 2023-01-04:
Hey all

I'll try and tackle a few things here. 

1. The migration to the new cert is **ONLY** supported with IoT Hub right now. DPS does not have the capability to use the Digicert only right now.
2. The expectation for the near future is that devices should have at least all **THREE** certs stored in them. That includes the Baltimore, Digicert Global, and Microsoft RSA.

As for this issue specifically, I think I found what the issue is. The documentation for the ESP32 transport doesn't say it directly, but I'm thinking that the call to `esp_transport_ssl_set_cert_data_der()` is for a single cert and not a chain or group of certs. So whichever cert is first is the one that will be used.

I have updated our code in a branch below which enables the use of a global cert store (aka not just for the single TLS session) that seems to be working for me with all **THREE** certs enabled and a hub with Digicert migration enabled. Please let me know if y'all are able to give it a try and if it works for you as well. Note that the change is in the `esp32` project only. So this will test basic connection and telemetry.

https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/dane/cacert-wip

sjpilot, 2023-01-05:
I'm running ADU example and it stopped working once switching the hub root certificate from Baltimore. I've manually merged the [https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/dane/cacert-wip](url) into the ADU example and it restored connectivity - I was getting a certificate error. 
I had to retain the #include "errno.h" as ADU example uses line 238 & 280: ESP_LOGE( TAG, "Reading failed, errno= %d", errno );

wreyford, 2023-01-05:
Hello @danewalton ,
I can confirm, that your fix using the global cert store, fixes the problem I experienced, and I'm sure you are correct in your interpretation of the `esp_transport_ssl_set_cert_data_der()` being for singular certificate, and that will explain very nicely what I experienced.
I have managed to provision a device using the chain of certificates, and also published telemetry data using a modified pnp code example.
Thx a lot.

danewalton, 2023-01-05:
Awesome thanks all for confirming. @sjpilot I added back the errno include.

We will look into productizing this update and have it merged in shortly. Thanks again for pointing it out.

PR #308 (2023-01-05): Update Root CA cert usage
Previously, the calls we were using with DER format would only add one cert. We need the whole group of CAs for future CA cert migration.

**NOTE** that SHA512 was required for mbedTLS configuration which includes SHA384 signature capabilities. The `Microsoft RSA Root Certificate Authority 2017` requires it.

Closes #299
closed 2023-01-05 by danewalton