Azure-Samples/iot-middleware-freertos-samples #303: System with a corrupted heap when using Azure DPS

tmaltesen, 2022-12-28:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>With the sample iot-middleware-freertos-samples\demos\projects\ESPRESSIF\adu
Configure the project to use DPS and setup a provisioning service on Azure
When connecting to the DPS server it will retrieve hub and device info, but right after crash with a CORRUPT HEAP
The problem seems to occur in the esp_transport_list_destroy
Some log messages below has been added by me!

### Any log messages given by the failure
>
I (39734) AZ IOT: Successfully acquired IoT Hub name and Device ID
I (39734) AZ IOT: Calling TLS_Socket_Disconnect
I (39754) esp_tls: Calling esp_transport_close
I (39754) esp_tls: Calling esp_transport_destroy
I (39754) esp_tls: Calling **esp_transport_list_destroy**
CORRUPT HEAP: multi_heap.c:108 detected at 0x3fca1bf4

abort() was called at PC 0x4038ef0d on core 0
0x4038ef0d: multi_heap_assert at D:/Espressif/frameworks/esp-idf-v4.4.3/components/heap/multi_heap_platform.h:61
 (inlined by) assert_valid_block at D:/Espressif/frameworks/esp-idf-v4.4.3/components/heap/multi_heap.c:106

Core  0 register dump:
MEPC    : 0x403808c0  RA      : 0x40389402  SP      : 0x3fcb1910  GP      : 0x3fc92600
0x403808c0: panic_abort at D:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_system/panic.c:402

0x40389402: __ubsan_include at D:/Espressif/frameworks/esp-idf-v4.4.3/components/esp_system/ubsan.c:294

TP      : 0x3fc726c8  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x3fcb1974  A0      : 0x3fcb193c  A1      : 0x3fcb1972
A2      : 0x00000000  A3      : 0x3fcb1969  A4      : 0x00000001  A5      : 0x3fc9e000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcb1a64  S3      : 0x3fcb1aa8
S4      : 0x3fcb1ab0  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Stack memory:
3fcb1910: 0x00000000 0x00000000 0x3fcb1970 0x4038f906 0x00000000 0x00000000 0x3fcb1ab0 0x3fc956a0
0x4038f906: abort at D:/Espressif/frameworks/esp-idf-v4.4.3/components/newlib/abort.c:43 (discriminator 3)

3fcb1930: 0x3fcb1974 0x3fc956bc 0x3fcb1970 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fcb1950: 0x43502074 0x34783020 0x65383330 0x20643066 0x63206e6f 0x2065726f 0x00000030 0x40380000
0x40380000: _vector_table at ??:?

3fcb1970: 0x00000030 0x38333034 0x64306665 0x00000000 0x3fcdc710 0x3fca1be0 0x3fca1bf4 0x4038ef10
0x4038ef10: multi_heap_get_allocated_size_impl at D:/Espressif/frameworks/esp-idf-v4.4.3/components/heap/multi_heap.c:118

3fcb1990: 0x00000000 0x3fca1bf4 0x3fca1be0 0x4038f012 0x00000000 0x3fcb0908 0x3fca1bf4 0x40380e22
0x4038f012: multi_heap_internal_lock at D:/Espressif/frameworks/esp-idf-v4.4.3/components/heap/multi_heap.c:152
 (inlined by) multi_heap_free_impl at D:/Espressif/frameworks/esp-idf-v4.4.3/components/heap/multi_heap.c:219

0x40380e22: heap_caps_free at D:/Espressif/frameworks/esp-idf-v4.4.3/components/heap/heap_caps.c:362

3fcb19b0: 0x3c145724 0x3fcb0908 0x3fcb0918 0x4038fa88 0x3fcb1a64 0x3fcb0420 0x3fcb0688 0x420afaa6
0x4038fa88: free at D:/Espressif/frameworks/esp-idf-v4.4.3/components/newlib/heap.c:40

0x420afaa6: esp_transport_destroy_foundation_transport at D:/Espressif/frameworks/esp-idf-v4.4.3/components/tcp_transport/transport_internal.c:36

3fcb19d0: 0x3fcb1a64 0x3fcb0420 0x3fcb0688 0x420af928 0x3fcb1aac 0x3fcb0420 0x3fcb0688 0x420aed42
0x420af928: base_destroy at D:/Espressif/frameworks/esp-idf-v4.4.3/components/tcp_transport/transport_ssl.c:289

0x420aed42: esp_transport_destroy at D:/Espressif/frameworks/esp-idf-v4.4.3/components/tcp_transport/transport.c:112

3fcb19f0: 0x000001f4 0x3fcb0420 0x00000000 0x420aed78 0x00000002 0x3fcb0420 0x3fcb0908 0x420aed98
0x420aed78: esp_transport_list_clean at D:/Espressif/frameworks/esp-idf-v4.4.3/components/tcp_transport/transport.c:86

0x420aed98: esp_transport_list_destroy at D:/Espressif/frameworks/esp-idf-v4.4.3/components/tcp_transport/transport.c:75

3fcb1a10: 0x3c145454 0x00009966 0x3c145000 0x42077f60 0x3fcb1aac 0x3fcb1aa4 0x3c145000 0x420764ac
0x42077f60: TLS_Socket_Disconnect at D:/Workspaces/PAQ7003/components/prolon-azure-iot/transport_tls_esp32.c:202 (discriminator 13)

0x420764ac: prvIoTHubInfoGet at D:/Workspaces/PAQ7003/components/prolon-azure-iot/prolon_azure_iot.c:1091

3fcb1a30: 0x3fc98e64 0x0000157c 0x4200847a 0x3fcb1a58 0x00000000 0x3c144000 0x3c145000 0x0000000a
0x4200847a: ullGetUnixTime at D:/Workspaces/PAQ7003/main/system_time.c:56

3fcb1a50: 0x0000003b 0x00000038 0x42078020 0x42078114 0x3fcb1a6c 0x00000000 0x3fcb0420 0x3fcb1a64
0x42078020: TLS_Socket_Recv at D:/Workspaces/PAQ7003/components/prolon-azure-iot/transport_tls_esp32.c:211

0x42078114: TLS_Socket_Send at D:/Workspaces/PAQ7003/components/prolon-azure-iot/transport_tls_esp32.c:253

3fcb1a70: 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x00000000 0x00000000 0x3c144000 0x3c145000 0x42076f38
0x42076f38: prvAzureTask at D:/Workspaces/PAQ7003/components/prolon-azure-iot/prolon_azure_iot.c:742

3fcb1a90: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0x00000000 0x00000000
3fcb1ab0: 0x00000000 0x4038c3c8 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
0x4038c3c8: prvTaskExitError at D:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/port/riscv/port.c:178

3fcb1ad0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1af0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc923a4 0x00000cb9 0x00000000
3fcb1b10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4038c3c8
0x4038c3c8: prvTaskExitError at D:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/port/riscv/port.c:178

3fcb1b30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb1b50: 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fcb1840 0x00000f50 0x3fc969c8 0x3fca55d8 0x3fcb1b5c
3fcb1b70: 0x3fc969c0 0x00000019 0x3fcb2474 0x3fcb2474 0x3fcb1b5c 0x00000000 0x00000000 0x3fcb0b58
3fcb1b90: 0x72757a41 0x73615465 0x0000006b 0x00000000 0x00000000 0x3fcb1b50 0x00000000 0x00000000
3fcb1bb0: 0x3fcb24a0 0x42000260 0x00000009 0x3fca2598 0x3fca2600 0x3fca2668 0x00000000 0x00000000
0x42000260: pthread_local_storage_thread_deleted_callback at D:/Espressif/frameworks/esp-idf-v4.4.3/components/pthread/pthread_local_storage.c:129

3fcb1bd0: 0x00000001 0x00000000 0x00000000 0x00000000 0x4210019e 0x00000000 0x00000000 0x00000000
0x4210019e: _cleanup_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/riscv32-esp-elf/src/newlib/newlib/libc/stdio/findfp.c:229

3fcb1bf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1c10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1c30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1c50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1c70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1c90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000
3fcb1cb0: 0x00000775 0x3fca1bf4 0x3fca1bf4 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1cd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb1cf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

### Expected/desired behavior
>
No crash ;-)

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Windows 11

### Versions
>

### Mention any other details that might be useful
Some other examples does not contain the TLS transport list, in these project there is an older bug causing trouble when reusing the TLS transport. The list seems to be a fix for this that was added to this sample but there is something wrong with the handling.
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

tmaltesen, 2022-12-28:
I assume that the call to esp_transport_list_destroy is trying to destroy an element that is already destroyed in the previous line, where esp_transport_destroy is called.

danewalton, 2022-12-28:
Hey @tmaltesen 
Thanks for the catch here. I think you're right. Here in the docs, it says the `_list_destroy()` will call the destroy for each of the transports. I'll look to fix it.

https://github.com/espressif/esp-idf/blob/a5b4e3bb3e1941710a33427b5d0816b7a429a63c/components/tcp_transport/include/esp_transport.h#L65

PR #304 (2023-01-09): only use list destroy
As noted in #303, the list destory will call the individual transport destroys.

https://github.com/espressif/esp-idf/blob/a5b4e3bb3e1941710a33427b5d0816b7a429a63c/components/tcp_transport/include/esp_transport.h#L65

closes #303
closed 2023-01-09 by danewalton