Azure-Samples/iot-middleware-freertos-samples #305: Update aziotkit and esp32 transport usage

danewalton, 2022-12-28:
The `aziotkit` and `esp32` samples both use the non "list" usage of the transports, which is an issue for ESP IDF v.4.4+. The ADU sample already uses the list idea. Update the other samples to conform.

ewertons, 2023-02-15:
@danewalton I see that both aziotkit and esp32 samples are working fine with ESP IDF v4.4.3. 
What is the issue of not having xTransportList in transport_tls_esp32.c for these samples?
Also, when I search for xTransportList in the adu sample, I see it created, populated, destroyed but never actually accessed. Could you help me understand where it is used? Thanks.

danewalton, 2023-02-21:
Yes this ultimately becomes a problem when there is an error on the transport. The error handling was changed with the new "list" way of using transports, and it was crashing when trying to access an error value which hadn't been malloc'd by the "list" struct (if I remember correctly). We don't access the list directly, but it's used within ESP TLS.

PR #349 (2023-03-02): update esp32 and aziot sample transport
closes #305
closed 2023-03-02 by danewalton