Azure-Samples/iot-middleware-freertos-samples #306: ADU esp32

Quinnvanderschaar2, 2023-01-03:
> Please provide us with the following information:
I don't succeed to do device updates on the esp32-c3. The device reboots when it tries to perform a update.
Last log message: I (46686) AZ IOT:  xStatus: 0
I have tried to search for the bug but I had to go down several layers.
I got stuck in the function sendHttpData at the pTransport->send( pTransport->pNetworkContext,....)  function call.

The latest released code in the main branch already crashes after making a connection. 
Last log: I (48854) AZ IOT: Successfully acquired IoT Hub name and Device ID
.
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>

### Any log messages given by the failure
>

### Expected/desired behavior
>

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Linux Mint
### Versions
> ESP-IDF v4.4.3-159-g54dcccc1f6

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

danewalton, 2023-01-03:
Hey @Quinnvanderschaar2 
I think this is related to this issue which I opened. 
https://github.com/Azure-Samples/iot-middleware-freertos-samples/issues/305
We just need to update the transport usage.

Do you mind trying the ADU sample and let us know if you still hit the same issue? It should contain the fix. Note that you don't to use the Device Update feature to run the sample. It should send telemetry.

Quinnvanderschaar2, 2023-01-03:
Thanks for the fast response, I already tried that before. Unfortunately these adjustments makes it worse. This results in that the devices crashes after the moment a connection was made to the iot hub. It doesn't even got the chance to do the update sequence. 

The last log is: I (50216) AZ IOT: Successfully acquired IoT Hub name and Device ID
CORRUPT HEAP: multi_heap.c:108 detected at 0x3fc9efc4

I have no problems with sending telemetry in general. The examples for that works overall fine, but I could not get any examples for the ADU working. I have tried the example in the main branch and the feature/iot-adu branch. Is there a branch available that has a stable version ( esp-idf version is not important for me) ?

danewalton, 2023-01-05:
Hey @Quinnvanderschaar2 
Firstly, the `feature/iot-adu` branch here and in the middleware source code was merged to main so that is stale now. I have now deleted it and `main` should be what you are working off of right now.

> I have no problems with sending telemetry in general. The examples for that works overall fine

Are you saying that you are aren't having any issues with the esp32 sample but you are with the ADU sample?

Quinnvanderschaar2, 2023-01-15:
Hi,
I have tried it now and the adu works now for single devices. Nevertheless, if I use de DPS it crashes now with the following error:

I (38397) AZ IOT: Successfully acquired IoT Hub name and Device ID
CORRUPT HEAP: multi_heap.c:108 detected at 0x3fc9e2c4

XavierGeerinck, 2023-01-24:
@danewalton could we have a call around this? Seeing that this is blocking our production deployment

vimeht, 2023-01-24:
@Quinnvanderschaar2, @XavierGeerinck can you please share the ADU agent and DPS logs from the device that has a repro so we can investigate on our side? It would be great if we can jump on a call this week too. Let me know what your schedule looks like

danewalton, 2023-01-27:
@Quinnvanderschaar2 I think this is resolved with the latest code changes. I'm going to close this for now but please let us know if you would like it reopened.

closed 2023-01-27 by danewalton