Azure-Samples/iot-middleware-freertos-samples #310: Getting component not ADU while trying the sample code.

fmadhurakkariyan, 2023-01-09:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Trying ADU update with ESP32 with the sample code

### Any log messages given by the failure
>AZ IOT: Properties component name: 
>AZ IOT: Component not ADU:
### Expected/desired behavior
>After receiving the writeable properties, the properties component name should be deviceUpdate 

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
Windows 10
### Versions
>ESPIDF 4.4.1

### Mention any other details that might be useful
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

danewalton, 2023-01-09:
Hi @fmadhurakkariyan 
Are you getting an empty component from the logs?

fmadhurakkariyan, 2023-01-10:
Hi @danewalton 
Yes, the logs show an empty component.

danewalton, 2023-01-10:
And this is an otherwise unaltered sample? You're running it straight from main?

fmadhurakkariyan, 2023-01-10:
It was a problem with the custom code. It is working now.

closed 2023-01-10 by fmadhurakkariyan