Azure-Samples/iot-middleware-freertos-samples #313: Crypto.h header file causes project not to build when enabling BLE MESH due to duplicate crypto.h file

lukecam95, 2023-01-10:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Enable BLE MESH in any project which is using this middleware project.

### Any log messages given by the failure
> ***/Cloud.c:1289:65: error: 'Crypto_HMAC' undeclared (first use in this function)

### Expected/desired behavior
> No errors when compiling.

### OS and Version?
> Windows 10 Pro.

### Versions
> 

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

danewalton, 2023-01-11:
Hi @lukecam95 
Which of our files is `crypto.h` and is causing a conflict? After a brief search, I can't seem to find one named that?

lukecam95, 2023-01-16:
For my version it was in path ..\demos\common\utilities. Maybe the files name was updated in more recent versions? Anyway the issue does not seem to exist on the latest branch.

closed 2023-01-16 by lukecam95