Azure-Samples/iot-middleware-freertos-samples #317: Timeout

Quinnvanderschaar2, 2023-01-15:
> Please provide us with the following information:
> ---------------------------------------------------------------

I am sending every minute telemetry via the esp32-c3 to the IOT-HUB.
Every time when the 67the message is send I get a timeout.
Error message:
E (3949864) TRANSPORT_BASE: poll_write select error 104, errno = Connection reset by peer, fd = 54
W (3949864) TRANSPORT_BASE: Poll timeout or error, errno=Connection already in progress, fd=54, timeout_ms=1107381270
E (3949884) tls_freertos: Writing failed, errno= 119
E (3949884) MQTT: Transport send failed. Error code=-1.
E (3949884) MQTT: Transport send failed for PUBLISH payload.
E (3949904) MQTT: MQTT PUBLISH failed with status MQTTSendFailed.
E (3949904) AZ IOT: Failed to publish telemetry: MQTT error=0x00000003
 

### This issue is for a: (mark with an `x`)
```
- [x ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>

### Any log messages given by the failure
>

### Expected/desired behavior
>

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

Linux Mint

### Versions
> 

ESP-IDF v4.4.3-159-g54dcccc1f6

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

danewalton, 2023-01-17:
Hey @Quinnvanderschaar2 
Which sample are you running and is it modified at all?

danewalton, 2023-01-27:
Closing this issue for now due to inactivity. If you would like it reopened, please let us know. For any other questions, please open another Github issue.

closed 2023-01-27 by danewalton