Azure-Samples/iot-middleware-freertos-samples #322: ullGetUnixTime() does not return an accurate Unix epoch in ST b-l475e-iot01a demo

warasilapm, 2023-01-20:
The implementation of ullGetUnixTime() in the ST b-l475e-iot01a demo project linked below does not return the actual epoch time.

https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/projects/ST/b-l475e-iot01a/main.c#L761-L777

This is due to the lack of any code updating `ulGlobalEntryTime`.

### This issue is for a: (mark with an `x`)
```
- [x ] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
Attempt to use the ST b-l475e-iot01a demo to connect over MQTT using a SAS token (though a symmetric key).

### Any log messages given by the failure
The log messages show a successful TLS connection but an unsuccessful MQTT connection where CONNACK indicates that the connection was refused because the request was not authorized. It should resemble the one shown below:
```
[INFO] [MQTT] [receivePacket:885] Packet received. ReceivedBytes=2.
[INFO] [MQTT] [deserializeConnack:970] CONNACK session present bit not set.
[ERROR] [MQTT] [logConnackResponse:917] Connection refused: not authorized.
[ERROR] [MQTT] [receiveConnack:1567] CONNACK recv failed with status = MQTTServerRefused.
[ERROR] [MQTT] [MQTT_Connect:1838] MQTT connection failed with status = MQTTServerRefused.
```

### Expected/desired behavior
The device connects and authenticates successfully.

### OS and Version?
This sample was compiled under Windows 10. I'm not sure this is relevant, since this is clearly an issue with the global never being updated beyond this past Sunday, January 15th, 2023.

### Versions
At a minimum the current version.

### Mention any other details that might be useful
This is a straightforward issue to resolve for some period of time into the future simply by providing a new global initializer far into the future. However, I'm surprised a library like coreSNTP wasn't leveraged to actually use the current epoch time.

danewalton, 2023-01-23:
Hey @warasilapm thanks for the message. We are aware of this and have an issue opened here tracking the updating of the samples: https://github.com/Azure-Samples/iot-middleware-freertos-samples/issues/133

We should be getting to this one soon.

warasilapm, 2023-01-23:
> Hey @warasilapm thanks for the message. We are aware of this and have an issue opened here tracking the updating of the samples: #133
> 
> We should be getting to this one soon.

Oh! My apologies for posting a duplicate, then. I didn't look that far back in the issue tracker. I only looked back a week or two since I figured this would be pretty major and obvious to everyone using the samples this past week.

Hopefully you'll have some time to look at it this year.

vaavva, 2023-02-07:
@warasilapm Until we have a more complete fix, I've pushed out the hard-coded time to unblock the samples in the linked PR

closed 2023-02-07 by vaavva