Azure-Samples/iot-middleware-freertos-samples #324: Error calling AzureIoTHubClient_SendPropertiesReported

afcec, 2023-01-26:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [X] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce

I am migrating my previous code from ESP-AZURE to Azure Iot Middleware for Freertos. I am using these samples as a reference for the developments and after some initial issues I am able to register the devices with the DPS and then connect successfully with the Iot Hub. 

As soon as the device connects to the hub I am trying to send reported properties updates but the call to AzureIoTHubClient_SendPropertiesReported always fail with return value eAzureIoTErrorPublishFailed. 

All the steps and code are exactly the same as in the examples. I am using self signed X.509 certificates for the devices.

I have tried with both, complex and simple messages with the same results.

If I do not call AzureIoTHubClient_SendPropertiesReported  the connection keeps established until the keep alive period expires. 

By the way, any call to AzureIoTHubClient_ProcessLoop causes an exception.

### Any log messages given by the failure

These are the logs I get:

E (106030) tls_freertos: Writing failed, errno= 0
E (106030) MQTT: Transport send failed. Error code=-1.
E (106030) MQTT: Transport send failed for PUBLISH header.
E (106110) MQTT: MQTT PUBLISH failed with status MQTTSendFailed.
E (106190) AZ IOT: Failed to Publish properties reported message: MQTT error=0x00000003

### Expected/desired behavior
>

### OS and Version?
> Windows 10 and ESP-IDF 4.4.3

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

afcec, 2023-01-26:
First attempt returns the log errors from above. Then if the device restarts the connection and again tries to send the reported properties there is an exception and decoding the backtrace it seems the source is in line 448 of transport_tls_esp32.c:

EspTlsTransportParams_t * pxEspTlsTransport = (EspTlsTransportParams_t *)pxTlsParams->xSSLContext;

danewalton, 2023-01-27:
Hi @afcec 
What is the exact payload which you are trying to send?

afcec, 2023-01-27:
I have tried with my original payload (can be quite large depending on the situation) but then I started testing just simple JSONs like this one with the same results:

"{ \"system.applianceType\": 2 }"

afcec, 2023-01-27:
Also the application crashes when I try to call AzureIoTHubClient_ProcessLoop...

afcec, 2023-01-27:
If I remove the call to AzureIoTHubClient_SendPropertiesReported and just leave AzureIoTHubClient_ProcessLoop the backtrace seems to point to the following line in transport_tls_esp32.c (line 414):

tlsStatus = esp_transport_read( pxEspTlsTransport->xTransport, pBuffer, xBytesToRecv, pxEspTlsTransport->ulReceiveTimeoutMs );

So once I am connected, it looks like no matter if I try to send or receive using the current socket, there is an exception.

danewalton, 2023-01-27:
What is your device ID just out of curiosity?

afcec, 2023-01-30:
I am using device ID 1901202300000001 for my last tests.

danewalton, 2023-01-30:
I'm getting some interesting logs looking at that device

```log
ErrorCode: DeviceNotFound
Message: Device 1901202300000001 at IotHub iot-hub-teka-smart-home not registered.
Timestamp: 01/19/2023 10:48:39
```

```log
Details: Client Id and Device Id in username do not match. If this is device module then client ID must be a Module ID; otherwise it must be a Device ID.
```

Are you sure you're setting the parameters correctly in the APIs?

afcec, 2023-01-31:
Hi Dane, those logs are quite old. If you see my description I confirm that the device registration is completed successfully and then I can stablish the connection with the IoT Hub. Only when I try to send/receive any data I am experiencing the issue.

That entry from 01/19 comes from a moment when I was also sending Module ID = Device ID in the API and then I found out that those parameters were not the same ( issue #298 ) 

Thanks

danewalton, 2023-02-01:
Are you able to send your client code so I can see if there's anything problematic there?

afcec, 2023-02-02:
Yes, of couse. I will send you an email later today with the code interacting with Azure along with some explanations, If you need anything else, just let me know. Thanks.

afcec, 2023-02-09:
Hi Dane, any news about this issue?

ewertons, 2023-02-16:
@afcec , if you got the code from https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/esp32 it should work on your own device. We run that sample frequently and the code is working for us.

I have a few questions in case you are still facing this issue in your application:
1. Did you run the sample above with your Azure IoT hub and device id?
2. What is the scale unit of your Azure IoT hub?
3. What is the specific ESP chip you are using in your tests (where the crash is happening)?

ewertons, 2023-02-18:
Hi @afcec , there will be an update for you through the support request you opened, but please find a reference sample we put together for you: https://github.com/ewertons/aziot-freertos-esp32 (1)

This sample (1) does call `AzureIoTHubClient_SendPropertiesReported` (https://github.com/ewertons/aziot-freertos-esp32/blob/main/components/sample-azure-iot/demos/sample_azure_iot/sample_azure_iot.c#L462) and `AzureIoTHubClient_ProcessLoop` (https://github.com/ewertons/aziot-freertos-esp32/blob/main/components/sample-azure-iot/demos/sample_azure_iot/sample_azure_iot.c#L453), showing these functions can be used without crashing the application.

We have used ESP IDF v.4.4.3 for validation.

afcec, 2023-02-20:
Hi @ewertons , thanks for your help and sorry for the delay in my reply. I was just trying to test again your previous sample because it has been a long time since I ran that code. Unfortunately, it is not so easy because I am using flash encryption and there are several additional tasks I need to accomplish:

- Changes in kconfig
- Modification of my original X.509 certificate and private key from PEM format to single line strings as requested by your sample (And make sure I do not make any mistake here…)
- Modifications in the code to enable operation with encrypted NVS

From my humble opinion, providing the certificate and private key from kconfig is not a good reference, you should try to create a more realistic sample. Let me make a few tests with the old and new code and I will get back to you ASAP.

afcec, 2023-02-20:
Hi again, I am having problems with the certificate/key as expected...

-  esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2780

I am doing the following: From my original X.509 files in PEM format I just replace "\r\n" with "\n". Please see the pictures attached (original PEM, modified removing   \r and kconfig) because there are no line feeds in the strings in kconfig. 

[image]
[image]
[image]

-

ewertons, 2023-02-20:
Hi @afcec ,
Please use the support issue you opened for sending sensitive data like certs or credentials, or logs with PII.
Apologies, but I took the freedom to redact your message and delete the certs info.

danewalton, 2023-02-23:
I believe this was tackled in the support request, and was related to calling middleware send/receive APIs in different threads. Which to note here is not natively supported.

Closing this for now but please let us know if you would like it reopened.

closed 2023-02-23 by danewalton

afcec, 2023-02-24:
Hi @danewalton , yes, I had not thought of it until yesterday and after changing the calls from one task to the other everything just worked fine.

Thank you all, your support was really appreciated.