Azure-Samples/iot-middleware-freertos-samples #368: FOTA with the ESP32 Documentation Inquiry

djalpiza, 2023-05-03:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [X] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Followed the tutorial available in the repo for "Perform an Over the Air Update with the ESP32". 

### Any log messages given by the failure
> The tutorial is working as expected. However I would like to get some guidance on how is it possible to change the "functional code" that is sent to the ESP32?

### Expected/desired behavior
> I would like to know if there is any additional documentation that help me understand from the available example of code: https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/adu

Where should I add my "functional code" for example I want to send an update to the ESP32 that not only show me the version of the FW that is running in console, but also is able to turn on the internal LED (FW v1.2) and another version of the FW that allows me to read information from a sensor (FW v1.3). 

To summarize I'd like to know how can I modify the available code in order to send along with the FOTA new functionalities to my ESP32 device?

The other thing I'd like to ask is related to the capability of utilizing non-volatile storage of device credentials, is there any example to code to follow to implement this?

### OS and Version?
> Windows 11

### Versions
>

### Mention any other details that might be useful

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

vaavva, 2023-05-03:
Hello @djalpiza!

Here's the loop in the sample where we're sending telemetry as well as checking for ADU updates - reading information from a sensor and sending it as telemetry would go well here: https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/sample_azure_iot_adu/sample_azure_iot_adu.c#L844

Above that is code that would run only once when the device is turned on (or restarted after an update), so that might be a better place to put something like printing the FW version so that it's not printing out repeatedly: https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/sample_azure_iot_adu/sample_azure_iot_adu.c#L704

Here's some information on using NVS for device credentials: https://github.com/Azure-Samples/iot-middleware-freertos-samples/discussions/243

djalpiza, 2023-05-04:
@vaavva thanks for the response. 

However I am still a little but lost on how to use this code as a baseline to program my ESP32 to fullfil my needs. 
What I am still trying to understand is how is this example: "iot-middleware-freertos-samples\demos\projects\ESPRESSIF\adu", which I have already tested and is working fine; is related to this code: "demos/sample_azure_iot_adu\sample_azure_iot_adu.c"?

What can I take as a baseline for my project? And where does my "special new function" is supposed to be placed in this already available structure for ADU?

vaavva, 2023-05-04:
@djalpiza  `demos/sample_azure_iot_adu\sample_azure_iot_adu.c` contains the sample code for ADU that all of the device samples can use, so this is where the code that interacts with the ADU service and IoT Hub goes (like connecting to IoT Hub, getting the update from ADU, etc).
`iot-middleware-freertos-samples\demos\projects\ESPRESSIF\adu` contains all of the code that is specific to just the ESP32 (like writing to flash, how to set the new image, and managing the WiFi).
You would use the code from both folders to create a baseline for your project.

Your new function could go in either place if you're only going to use your sample for the ESP32, since then there wouldn't be a problem with using functionality specific to the board in the generic ADU code. Placing an ESP32 specific function in the ESP32 folder might be easier though since it already includes a lot of the device files.

djalpiza, 2023-05-09:
OK, I can see that the code: \iot-middleware-freertos-samples\demos\projects\ESPRESSIF\adu\main\**azure_iot_freertos_esp32_main.c**

Calls this function:  vStartDemoTask();

Which is defined on: \iot-middleware-freertos-samples\demos\sample_azure_iot_adu\sample_azure_iot_adu.c
And has the following sub functions: 
void vStartDemoTask( void )
{
    /* This example uses a single application task, which in turn is used to
     * connect, subscribe, publish, unsubscribe and disconnect from the IoT Hub */
    xTaskCreate( prvAzureDemoTask,         /* Function that implements the task. */
                 "AzureDemoTask",          /* Text name for the task - only used for debugging. */
                 democonfigDEMO_STACKSIZE, /* Size of stack (in words, not bytes) to allocate for the task. */
                 NULL,                     /* Task parameter - not used in this case. */
                 tskIDLE_PRIORITY,         /* Task priority, must be between 0 and configMAX_PRIORITIES - 1. */
                 NULL );                   /* Used to pass out a handle to the created task - not used in this case. */
}

If I understood correctly, this function can not be removed from the program since it is the general function to connect to the IoT Hub, get the Updates for ADU, etc, is this right? 

If I just want to add a simple function to this code that allows me to change the blinking frequency of the ESP32 internal LED, named: "Blink", where should this function be defined?

djalpiza, 2023-05-09:
My question comes because this function will be an infinite loop, turning ON and OFF and LED (simple example of a function that the ESP32 can perform) but I still want to have the ability to send OTA updates of the FW to my device. 

But honestly I don't understand where this function"Blink" should be defined and called. Can it be integrated to this available example for ADU? Or is it better to have an specific void app_main( void ) that uses the ADU example (azure_iot_freertos_esp32_main.c)?

vaavva, 2023-05-09:
@djalpiza Yes, the functionality of `vStartDemoTask` and `prvAzureDemoTask`, which it calls (and has the connection and ADU functionality) cannot be removed if you want to connect to IoT Hub and use ADU.

A Blink function can definitely be integrated into this example for ADU! The easiest way would be to create a second FreeRTOS task (in addition to the [demo task](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/sample_azure_iot_adu/sample_azure_iot_adu.c#L1167)) with `xTaskCreate` that contains your Blink function. You could initialize it with the demo task or within `app_main` in the [ESP main.c file](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/projects/ESPRESSIF/adu/main/azure_iot_freertos_esp32_main.c#L327). It would probably be easiest to define the task itself within `azure_iot_freertos_esp32_main.c` or the `ESPRESSIF/adu` folder since it already has ESP32 `#include`s

djalpiza, 2023-05-10:
Thanks @vaavva .

I have one more question regarding how the updates that are uploaded to Azure with the manifest are assigned to different groups? 

I am using this command to generate the manifest: 

az iot du update init v5 --update-provider djalpiza --update-name blink --update-version 1.1 --compat deviceModel=ESP32-Azure-IoT-Kit deviceManufacturer=ESPRESSIF --step handler=microsoft/swupdate:1 properties='{\"installedCriteria\":\"1.1\"}' --file path=./blink_esp32-v1.1.bin > ./djalpiza.blink.1.1.importmanifest.json

As per my understanding, these parameters will determine to which groups the update can be deployed: 
- deviceModel
- deviceManufacturer
Is there any way to specify that a certain response is targeted to a specific tag? For example, all devices that are tagged to "ADUGroup": "T2", BUT devices tagged to: "ADUGroup": "T1" do not get the new Update FW as available to them?

If I check my Devices twin, I cn see that it matches:
[image]

Do these values are populated automatically when the device is connected to Azure? Or can they be manually modified?
I'd like to start working now with other platforms like Raspberry Pi, Azure Development Kits, and others. How can I set the update to be targeted only for the specific device it is intended to? 

Also, something I am wondering is, if there is any way to rollback to a FW version that is previously installed as an update?
For example: 
1- I initialize my device on FW v1.0
2- I upload the FWv1.0 in Azure as an Update
3- I update my device (with OTA) to FW v1.1
4- I update my device (with OTA0 to FW v1.2
5- After a while I want to go back all the way to FW v1.0. 

In this scenario, can I simply choose this FW to rollback? Or do I need to upload it once again to Azure to send it to my devices via OTA?

vaavva, 2023-05-10:
@djalpiza Split this out into a few answers:
> Is there any way to specify that a certain response is targeted to a specific tag? For example, all devices that are tagged to "ADUGroup": "T2", BUT devices tagged to: "ADUGroup": "T1" do not get the new Update FW as available to them?

Any device that has a matching manufacturer and model will have the update available to it, but if you have them tagged with different groups, you could deploy it to only certain groups. If you have devices that would never get the same updates (i.e. if you have some ESP32s that you have acting as temperature sensors and some acting as LED controllers, and they'd never get the same firmware), then you could give them different values for one or both of these fields (described below)

> Do these values are populated automatically when the device is connected to Azure? Or can they be manually modified?

These values are set for the device in the demo config (https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/projects/ESPRESSIF/adu/config/demo_config.h#L228-L232). You can see an example of setting these values for a different device in our ST L475 sample: https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/main/demos/projects/ST/b-l475e-iot01a/config/demo_config.h#L259-L263
If you make changes here, remember to make the same changes in the manifest generation command.
> In this scenario, can I simply choose this FW to rollback? Or do I need to upload it once again to Azure to send it to my devices via OTA?

Yep! If you go in the portal into `Updates` -> `Groups and Deployments` -> your update group, you can go into `Current updates` -> click `View available updates`, and select `Deploy` for an older version
[image]

djalpiza, 2023-05-17:
Hi @vaavva, this might be a little off from my initial inquiry. However, I am now trying to enable OTA for a Raspberry Pi 3+. I am following the tutorial available here: https://learn.microsoft.com/en-us/azure/iot-hub-device-update/device-update-raspberry-pi

I am able to follow the tutorial, and get the device conected and ready to recieve updates from Azure IoT Hub. My question, is similar to what I wanted to to with the ESP32 Blink Function. 

Taking the file from: https://github.com/Azure/iot-hub-device-update/releases. Can you guide me on what has to be modifed in order to set new functionalities to the updated image in the Raspberry Pi? Keeping the connectivity and availablity for new updates from the Azure IoT Hub.

vaavva, 2023-05-18:
@djalpiza since the raspberry pi sample is in a different repo, please open an issue with them there - feel free to link this issue for context.

vaavva, 2023-06-01:
Closing this for now as I think the questions related to this repo are resolved, but please let us know if you would like it reopened.

closed 2023-06-01 by vaavva