Azure-Samples/iot-middleware-freertos-samples #369: Potential stack corruption in sockets_wrapper_lwip.c

glenn-edna, 2023-05-24:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
> Set `lwipdnsresolverMAX_WAIT_SECONDS` shorter than it takes for the dns request to complete.

### Any log messages given by the failure
> No

### Expected/desired behavior
> No stack corruption 😉 

### OS and Version?
> Related to lwip example code, not os specific

### Versions
> main branch

### Mention any other details that might be useful

In the `prvGetHostByName` method, `dns_gethostbyname_addrtype` is called with a local variable `ulAddr` in which the result of the dns query is stored by the `lwip_dns_found_callback`. In the case the lookup for the host happens for the first time (ie it is not yet cached and the gethostbyname call thus returns `ERR_INPROGRESS`), the `prvGetHostByName` method blocks until either `ulAddr` is no longer zero, or a timeout occurs.

There are two potential issues with this implementation:

* When the dns query fails (ie the hostname could not be resolved / a timeout happens), the callback sets `ulAddr` to `0` which cannot be distinguished from the lookup still being in progress. The `prvGetHostByName` will therefore keep blocking until 20s (timeout in the example code) have passed, even though the request has long finished.
* When the dns query takes longer than the timeout (which is set to 20s in the example, but could be set much shorter in production code), the `prvGetHostByName` function exists with return code 0, popping `ulAddr` from the stack. But the dns lookup keeps running in the background and invokes `lwip_dns_found_callback` when the lookup completes (or fails). When the happens, it attempts to write the ip address (or 0) to `ulAddr` which no longer exists, overwriting whatever is now living in that spot on the stack.

Possible solution:

Instead of just passing `ulAddr` to the callback, pass a struct that also contains a semaphore. The `prvGetHostByName` can then create the semaphore on the stack, wait for it in `ERR_INPROGRESS` instead of using a loop with a timeout, and delete the semaphore again at the end of the function. The callback `lwip_dns_found_callback` can then set the ip address in the struct and give the semaphore, signaling to the `prvGetHostByName` that the lookup attempt finished.

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

RLeclair, 2023-06-09:
Hi @glenn-edna, thank you for reporting this bug! I will be looking into it and trying to reproduce it, thank you for the detailed explanation of the issue.

RLeclair, 2023-06-15:
PR with fix has been closed, thank you for pointing this bug out @glenn-edna

closed 2023-06-15 by RLeclair