Azure-Samples/iot-middleware-freertos-samples #370: Can anyone help to use vscode to build example

JamesJian-tech, 2023-05-27:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [x] bug report -> please search issues before submitting
- [ ] feature request
- [ ] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>

### Any log messages given by the failure
>
[image]
[1084/1109] Building C object esp-idf/azure-iot-middleware-freertos/CMakeFiles/__idf_azure-iot-middleware-freertos.dir/b587fa35c5ad01fd19af7f6f470063a1/azure_iot.c.obj
FAILED: esp-idf/azure-iot-middleware-freertos/CMakeFiles/__idf_azure-iot-middleware-freertos.dir/b587fa35c5ad01fd19af7f6f470063a1/azure_iot.c.obj

### Expected/desired behavior
>
I am using vscode to build the example, I got the error message : 

[1084/1109] Building C object esp-idf/azure-iot-middleware-freertos/CMakeFiles/__idf_azure-iot-middleware-freertos.dir/b587fa35c5ad01fd19af7f6f470063a1/azure_iot.c.obj
FAILED: esp-idf/azure-iot-middleware-freertos/CMakeFiles/__idf_azure-iot-middleware-freertos.dir/b587fa35c5ad01fd19af7f6f470063a1/azure_iot.c.obj

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)

### Versions
>
windows 11
IDF 4.4.3

### Mention any other details that might be useful

I can successfully build the example by using ESP-IDF powershell.
[image]

> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

vaavva, 2023-05-30:
Hello @JamesJian-tech! Currently we only support using the ESP-IDF to build the samples. Are you using the command line when you try and build the sample in vscode, or are you using an extension?

JamesJian-tech, 2023-06-04:
Yes, I have fixed now all good, thanks

closed 2023-06-04 by JamesJian-tech