Azure-Samples/iot-middleware-freertos-samples #381: How to receive desired properties from IoT hub from device twin

amolchor20, 2023-10-26:
> Please provide us with the following information:
> ---------------------------------------------------------------

### This issue is for a: (mark with an `x`)
```
- [ ] bug report -> please search issues before submitting
- [ ] feature request
- [x] documentation issue or request
- [ ] regression (a behavior that used to work and stopped in a new release)
```

### Minimal steps to reproduce
>1. update the device twin properties from azure IoT hub.
2. wait on device side to receive the updated desired properties from IoT hub.
3. there is no debug print for payload indicate that payload received from IoT hub for the change of desired properties. 

### Any log messages given by the failure
>No logs for desired properties from IoT hub.

### Expected/desired behavior
>Get the debug print of desired properties as payload in callback function/simple 
debug print of all the properties from device twin.

### OS and Version?
> Windows 7, 8 or 10. Linux (which distribution). macOS (Yosemite? El Capitan? Sierra?)
freeRTOS

### Versions
>azure middleware version 1.1.0

### Mention any other details that might be useful
I am using azure iot middleware on stm32u585 (B-U585I-IOT02A) board.
I am able to send telemetry messages to IoT devices and also cloud to device.
when I am changing the desired properties from IoT hub,
how can I verify the desired properties are changed from device side?
> ---------------------------------------------------------------
> Thanks! We'll be in touch soon.

NikaChikovani, 2023-12-16:
I think first of all you have to subscribe to the properties topic in order to get notifications about the update. Use function called AzureIoTHubClient_SubscribeProperties() after connecting to the IoT Hub.

ericwol-msft, 2023-12-19:
Please look at sample [aziotkit](https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/projects/ESPRESSIF/aziotkit) for ESP32.

[vHandleWritableProperties](https://github.com/Azure-Samples/iot-middleware-freertos-samples/blob/794c4936e6573baa636f44f78d8e3878cd49fdfa/demos/projects/ESPRESSIF/aziotkit/main/azure_iot_freertos_esp32_sensors_data.c#L380C6-L380C31) is called for twin property updates.

Output should look as follows:

> I (24569) MQTT: Packet received. ReceivedBytes=3470.
> I (24569) MQTT: De-serialized incoming PUBLISH packet: DeserializerResult=MQTTSuccess.
> I (24569) MQTT: State record updated. New state=MQTTPublishDone.
> I (24579) AZ IOT: $iothub/twin/PATCH/properties/desired/?$version=24
> I (24599) sample_azureiotkit: Telemetry frequency set to once every 10 seconds.
> I (24619) AZ IOT: Successfully parsed properties

amolchor20, 2024-01-19:
Thanks for reply !!

currently my board stm32u585 is connected with cellular board for internet connectivity.
And I have integrated [adu ](https://github.com/Azure-Samples/iot-middleware-freertos-samples/tree/main/demos/sample_azure_iot_adu) code from middleware.
for device update the desired properties is changed when i deploy the update from Hub.
[image]
but the changed desired property is not received on device side.
Although the freeRTOS middleware code is using the 
[image]
Api shown in Image.
Could you please guide me on regarding desired properties update?

Thanks in advance !!

amolchor20, 2024-01-22:
Hii,

I have enabled debug logs in coremqtt file.
I found below logs for the API
            /* Get property document after initial connection */
            `xResult = AzureIoTHubClient_RequestPropertiesAsync( &xAzureIoTHubClient );`
[image]

**logs :**
[2024-01-22_10:55:39:016]$$$$$$$$$$$$$$$$ lets start the app 

[2024-01-22_10:55:39:016]===== X-Cube-Cellular version : X-CUBE-CELLULAR-7.1.0 =====

[2024-01-22_10:55:39:016]CST: New State: BOOT_STATE

[2024-01-22_10:55:39:016]BG96 UART config: BaudRate=115200 / HW flow ctrl=1

[2024-01-22_10:55:39:016]CST: osCDS_cellular_service_init()

[2024-01-22_10:55:39:016]CST: modem state : CA_MODEM_POWER_OFF
[2024-01-22_10:55:39:040]
[2024-01-22_10:55:39:040]CST: CST_notif_callback (Data Cache event=0)

[2024-01-22_10:55:39:040]CST: CST_notif_callback (Data Cache event=3)

[2024-01-22_10:55:39:040]CST: CST_send_message: cmd 0

[2024-01-22_10:55:39:040]CST: AUTOM TASK: BOOT_STATE - BOOT_EVENT

[2024-01-22_10:55:39:040]CST: CST_boot_event_mngt

[2024-01-22_10:55:39:040]CST: New State: MODEM_INIT_STATE

[2024-01-22_10:55:39:040]CST: CST_send_message: MODEM_INIT_EVENT

[2024-01-22_10:55:39:084]CST: AUTOM TASK: MODEM_INIT_STATE - MODEM_INIT_EVENT

[2024-01-22_10:55:39:084]CST: CST_init_state_mngt

[2024-01-22_10:55:39:084]CST: osCDS_power_on()

[2024-01-22_10:55:39:084][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:709] ------------------------------------------------------------------------------
[2024-01-22_10:55:39:084][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:710] ADU SAMPLE
[2024-01-22_10:55:39:119][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:711] Version: 0.0
[2024-01-22_10:55:39:119][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:712] ------------------------------------------------------------------------------
[2024-01-22_10:55:39:119][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:1139] Creating a TLS connection to nimazureiothub.azure-devices.net:8883.
[2024-01-22_10:55:41:331]SysCtrl_BG96:Waiting 5500 millisec for modem running...
[2024-01-22_10:55:46:818]
[2024-01-22_10:55:46:818]SysCtrl_BG96:...done

[2024-01-22_10:55:46:818]    ATParser:*** SEND (size=11) ***

[2024-01-22_10:55:46:818]AT+IFC=2,2<CR>

[2024-01-22_10:55:46:818]    +QIURC: "recv",1<CR><LF>

[2024-01-22_10:55:46:818]    <CR><LF>

[2024-01-22_10:55:46:818]    OK<CR><LF>

[2024-01-22_10:55:46:818]    ATParser:*** SEND (size=3) ***

[2024-01-22_10:55:46:818]AT<CR>

[2024-01-22_10:55:46:818]    <CR><LF>

[2024-01-22_10:55:46:818]    OK<CR><LF>

[2024-01-22_10:55:46:818]BG96:modem synchro established, proceed to normal power sequence

[... 2373 lines cut ...]

[2024-01-22_10:57:53:536]AT+QISEND=1,85<CR>
[2024-01-22_10:57:53:560]
[2024-01-22_10:57:53:560]    <CR><LF>

[2024-01-22_10:57:53:560]    ERROR<CR><LF>

[2024-01-22_10:57:53:560]ATCore ERROR:AT_sendcmd error

[2024-01-22_10:57:53:560]ATCore ERROR:AT_sendcmd error: parse from rsp

[2024-01-22_10:57:53:560]LOG ERROR #3: channel=1 / errorId=11 / gravity=2

[2024-01-22_10:57:53:560]LOG ERROR #4: channel=1 / errorId=14 / gravity=2

[2024-01-22_10:57:53:560]CS ERROR:<Cellular_Service> error when sending data to socket

[2024-01-22_10:57:53:583]ComLib ERROR: snd data NOK at low level

[2024-01-22_10:57:53:583][DEBUG] [MQTT] [core_mqtt.c:643] BytesSent=28, BytesRemaining=0
[2024-01-22_10:57:53:583][DEBUG] [MQTT] [core_mqtt.c:665] Successfully sent packet at time 19159.
[2024-01-22_10:57:53:583][DEBUG] [MQTT] [core_mqtt.c:1419] Sent 28 bytes of PUBLISH header.
**[2024-01-22_11:00:25:727][DEBUG] [MQTT] [core_mqtt.c:1443] PUBLISH payload was not sent. Payload length was zero.**

Is there any error related to **AzureIoTHubClient_RequestPropertiesAsync** api that why I didn't get the device twin desired properties when it is updated?

amolchor25, 2024-01-24:
Below is the logs from device boot up to `AzureIoTHubClient_RequestPropertiesAsync()`
[image]

logs : 
_[2024-01-22_11:52:53:579]$$$$$$$$$$$$$$$$ lets start the app 

[2024-01-22_11:52:53:579]===== X-Cube-Cellular version : X-CUBE-CELLULAR-7.1.0 =====

[2024-01-22_11:52:53:579]CST: New State: BOOT_STATE

[2024-01-22_11:52:53:579]BG96 UART config: BaudRate=115200 / HW flow ctrl=1

[2024-01-22_11:52:53:579]CST: osCDS_cellular_service_init()

[2024-01-22_11:52:53:579]CST: modem state : CA_MODEM_POWER_OFF
[2024-01-22_11:52:53:592]
[2024-01-22_11:52:53:592]CST: CST_notif_callback (Data Cache event=0)

[2024-01-22_11:52:53:592]CST: CST_notif_callback (Data Cache event=3)

[2024-01-22_11:52:53:592]CST: CST_send_message: cmd 0
[2024-01-22_11:52:53:614]
[2024-01-22_11:52:53:614]CST: AUTOM TASK: BOOT_STATE - BOOT_EVENT

[2024-01-22_11:52:53:614]CST: CST_boot_event_mngt

[2024-01-22_11:52:53:614]CST: New State: MODEM_INIT_STATE

[2024-01-22_11:52:53:614]CST: CST_send_message: MODEM_INIT_EVENT

[2024-01-22_11:52:53:614]CST: AUTOM TASK: MODEM_INIT_STATE - MODEM_INIT_EVENT

[2024-01-22_11:52:53:614]CST: CST_init_state_mngt

[2024-01-22_11:52:53:614]CST: osCDS_power_on()

[2024-01-22_11:52:53:614][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:709] ------------------------------------------------------------------------------
[2024-01-22_11:52:53:637][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:710] ADU SAMPLE
[2024-01-22_11:52:53:637][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:711] Version: 1.0
[2024-01-22_11:52:53:637][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:712] ------------------------------------------------------------------------------
[2024-01-22_11:52:53:661][INFO] [AzureIoTDemo] [sample_azure_iot_adu.c:1139] Creating a TLS connection to nimazureiothub.azure-devices.net:8883.
[2024-01-22_11:52:55:893]SysCtrl_BG96:Waiting 5500 millisec for modem running...
[2024-01-22_11:53:01:362]
[2024-01-22_11:53:01:362]SysCtrl_BG96:...done

[2024-01-22_11:53:01:362]    ATParser:*** SEND (size=11) ***

[2024-01-22_11:53:01:362]AT+IFC=2,2<CR>
[2024-01-22_11:53:01:393]
[2024-01-22_11:53:01:393]    <CR><LF>

[2024-01-22_11:53:01:393]    OK<CR><LF>

[2024-01-22_11:53:01:393]    ATParser:*** SEND (size=3) ***

[2024-01-22_11:53:01:393]AT<CR>

[2024-01-22_11:53:01:393]    <CR><LF>

[2024-01-22_11:53:01:393]    OK<CR><LF>

[2024-01-22_11:53:01:393]BG96:modem synchro established, proceed to normal power sequence

[2024-01-22_11:53:01:393]        ATParser:*** SEND (size=5) ***

[2024-01-22_11:53:01:393]ATE0<CR>

[2024-01-22_11:53:01:393]    <CR><LF>

[2024-01-22_11:53:01:393]    OK<CR><LF>
[... 2355 lines cut ...]
[2024-01-22_11:53:19:969]    ATParser:*** SEND (size=15) ***

[2024-01-22_11:53:19:969]AT+QISEND=1,85<CR>
[2024-01-22_11:53:19:986]
[2024-01-22_11:53:19:986]    <CR><LF>

[2024-01-22_11:53:19:986]    > 

[2024-01-22_11:53:19:986]BG96: SOCKET PROMPT RECEIVED

[2024-01-22_11:53:19:986]        ATParser:*** SEND (size=85) ***

[2024-01-22_11:53:19:986]ATParser: Big frame (display deactivated) 
[2024-01-22_11:53:20:016]
[2024-01-22_11:53:20:016]    <CR><LF>

[2024-01-22_11:53:20:016]    SEND OK<CR><LF>

[2024-01-22_11:53:20:016]ComLib: snd data ok

[2024-01-22_11:53:20:016][DEBUG] [MQTT] [core_mqtt.c:643] BytesSent=28, BytesRemaining=0
[2024-01-22_11:53:20:016][DEBUG] [MQTT] [core_mqtt.c:665] Successfully sent packet at time 19236.
[2024-01-22_11:53:20:016][DEBUG] [MQTT] [core_mqtt.c:1419] Sent 28 bytes of PUBLISH header.
[2024-01-22_11:53:22:164][DEBUG] [MQTT] [core_mqtt.c:1443] PUBLISH payload was not sent. Payload length was zero.
_

amolchor25, 2024-01-30:
Why I am not receiving the device twin notification, because of **"AzureIoTHubClient_ProcessLoop"** function.

If there is an update notification present from Azure IoT hub then the  **"AzureIoTHubClient_ProcessLoop"**  function  gives device twin notification, **when there is no notification from Azure IoT Hub then it will throw below error**

[image]

and another error is
 
[image]

because of above two error I have removed the  **"AzureIoTHubClient_ProcessLoop"**  function from my code.

these two errors are reproduce constantly.